Botspot/bvm #1: Stuck at firstboot

gitbls, 2025-02-26:
Running `bvm firstboot` I get:
```
bls@pit~> ./bvm/bvm firstboot ~/win11
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██                                                      
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███                                                      
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█                                                      
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█                                                      
 █████🭠   🭖██🭡   █▉     🮋█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"Making it easy to run a Windows 11 VM on that potato you found in the woods."
BVM mode firstboot
Windows should install 100% automatically. This will take several hours.
qemu-system-aarch64: Display 'gtk' is not available. Perhaps you want to install qemu-system-gui package?
```
I tried installing qemu-system-gui:
```
apt install qemu-system-gui
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 qemu-system-gui : Depends: qemu-system-x86 (= 1:7.2+dfsg-7+deb12u12) or
                            qemu-system-arm (= 1:7.2+dfsg-7+deb12u12) or
                            qemu-system-mips (= 1:7.2+dfsg-7+deb12u12) or
                            qemu-system-ppc (= 1:7.2+dfsg-7+deb12u12) or
                            qemu-system-sparc (= 1:7.2+dfsg-7+deb12u12) or
                            qemu-system-misc (= 1:7.2+dfsg-7+deb12u12)
E: Unable to correct problems, you have held broken packages.
```
This is on a freshly installed 2024-11-19 Desktop IMG. The steps prior to firstboot all completed without error.

Thx!

closed 2025-02-26 by gitbls

Botspot, 2025-02-26:
Are you using Pi OS 32-bit? I did not need to install qemu-system-gui on my system, and from a quick search it only appears to be available for armhf.

gitbls, 2025-02-26:
No, this is a Pi5 with a freshly imaged 64bit Desktop version.