Botspot/bvm #2: Failed to find romfile efi-virtio.rom

gitbls, 2025-02-26:
`./bvm/bvm firstboot ~/win11` gets me
```
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██                                                      
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███                                                      
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█                                                      
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█                                                      
 █████🭠   🭖██🭡   █▉     🮋█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"With BVM you get the best of both worlds... or at least bragging rights."
BVM mode firstboot
Windows should install 100% automatically. This will take several hours.
qemu-system-aarch64: -device virtio-net-pci,netdev=nic: failed to find romfile "efi-virtio.rom"
QEMU did not exit successfully.
```
I don't see the romfile anywhere on my disk, and it didn't seem to be hiding in virtio-win.iso

Is this an installer issue or my pebcak?

Botspot, 2025-02-26:
It's likely a missing dependency. Try `sudo apt install ipxe-qemu`
If that works I'll add it to the list of dependencies.

gitbls, 2025-02-26:
Yes, that did it. I added the one from bookworm-backports and am now starting Win11 setup. Very cool!

Botspot, 2025-02-26:
Fixed in commit https://github.com/Botspot/bvm/commit/d2b61e11c9eb278f2162639875b54b746fa220eb

closed 2025-02-26 by Botspot