Botspot/bvm #3: Small speed bumps along the way

gitbls, 2025-02-26:
When firstboot completed, it issued the following output:
```
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██                                                      
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███                                                      
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█                                                      
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█                                                      
 █████🭠   🭖██🭡   █▉     🮋█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"Because there is no true difference between ‘unsupported’ and ‘fun challenge.’"
BVM mode firstboot
Windows should install 100% automatically. This will take several hours.
QEMU closed.
qemu-nbd: Failed to blk_new_open 'firstboot': Could not open 'firstboot': No such file or directory
/dev/nbd0 disconnected
mount_qcow2: failed to connect firstboot to nbd device for mounting
unmount_qcow2: 'qemu-nbd: Cannot open /dev/: Is a directory'
It appears that Windows installation was not completed in disk.qcow2
```
Nonetheless I forged ahead and did a 'bvm boot'. The first time I ran it, it failed:
```
bvm/bvm boot ~/win11
 █████߭ ߭█߭  ߭█߭ ██◣   ◢██                                                      
 █▊  ߭█ ߭█߭  ߭█߭ ███◣ ◢███                                                      
 █████߭  ߭█߭߭█߭  █▉◥█߭█◤߮█                                                      
 █▊  ߭█  ߭█߭߭█߭  █▉ ◥█◤ ߮█                                                      
 █████߭   ߭██߭   █▉     ߮█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"PROTIP: Try selecting the lines of text above for an easter egg! :)"
Auto-updating BVM for the latest features and improvements...
remote: Enumerating objects: 8, done.
remote: Counting objects: 100% (8/8), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 6 (delta 2), reused 6 (delta 2), pack-reused 0 (from 0)
Unpacking objects: 100% (6/6), 1.02 KiB | 174.00 KiB/s, done.
From https://github.com/Botspot/bvm
   71a0b3f..acb672d  main       -> origin/main
Updating 71a0b3f..acb672d
Fast-forward
 bvm | 20 ++++++++++++++------
 1 file changed, 14 insertions(+), 6 deletions(-)
git pull finished. Reloading script...
Must specify a mode
 █████߭ ߭█߭  ߭█߭ ██◣   ◢██                                                      
 █▊  ߭█ ߭█߭  ߭█߭ ███◣ ◢███                                                      
 █████߭  ߭█߭߭█߭  █▉◥█߭█◤߮█                                                      
 █▊  ߭█  ߭█߭߭█߭  █▉ ◥█◤ ߮█                                                      
 █████߭   ߭██߭   █▉     ߮█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"Rest easy knowing you are better than every Windows on Raspberry user."
BVM mode help
BVM HELP:   --------------------------------------------------------------------
All primary modes: new-vm, download, prepare, firstboot, boot, connect, mount
You generally want to run the modes in that order.
Additional modes: help, list-languages, boot-nodisplay, boot-ramfb, boot-gtk, connect-freerdp, connect-remmina

To get a fresh VM up and running, use a sequence like this:
~/bvm/bvm new-vm ~/win11
 This makes a config file: ~/win11/bvm-config   <--- Please read it!
~/bvm/bvm download ~/win11
 This downloads Windows and necessary drivers.
~/bvm/bvm prepare ~/win11
 This bundles everything up to get ready for first boot.
~/bvm/bvm firstboot ~/win11
 If the Windows install is interrupted, just run this step again.
 Be aware: when Windows finishes installing, the VM will shutdown.
 All .iso files and the unattended folder could be removed once this step is done.
~/bvm/bvm boot ~/win11
 Main command to use the VM.
This works, but is a little laggy and lacks crucial features.
 It is recommended to boot the VM headless and then connect to it via RDP:
~/bvm/bvm boot-nodisplay ~/win11
 Then in a second shell, connect to the RDP service:
~/bvm/bvm connect ~/win11
 The connect mode has better audio, clipboard sync, file sharing, dynamic screen resizing, and a higher frame rate.
 Default connect mode uses the FreeRDP client, which is faster, but sometimes freezes on login. If this happens try the connect-remmina mode.
 Mount the Windows main hard drive with:
~/bvm/bvm mount ~/win11
 Be aware: the VM needs to be shut down properly to mount in read/write mode.
```
But it worked correctly when I reran it, and Win11 is now running on my desktop!
```
bvm/bvm boot ~/win11
 █████߭ ߭█߭  ߭█߭ ██◣   ◢██                                                      
 █▊  ߭█ ߭█߭  ߭█߭ ███◣ ◢███                                                      
 █████߭  ߭█߭߭█߭  █▉◥█߭█◤߮█                                                      
 █▊  ߭█  ߭█߭߭█߭  █▉ ◥█◤ ߮█                                                      
 █████߭   ߭██߭   █▉     ߮█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"Achievement unlocked: Running Windows where it absolutely should not be."
BVM mode boot
```
I'm already seeing a wishlist (#1: bigger window!), but I get that most important now is to get the install process working reliably and repeatably. How can I help?

Botspot, 2025-02-26:
You encountered 2 issues:
1. Some kind of issue with the Windows Defender removal. (likely an easy fix, and just from thinking through it I think I know what went wrong)
2. One-time issue with how the script restarts itself after receiving an update. Already fixed in commit https://github.com/Botspot/bvm/commit/d2b61e11c9eb278f2162639875b54b746fa220eb, but anyone who cloned the repo before that will encounter this one single time.

How to help? I can think of a few ways:

1. Testing on system setups that I don't have. For example, a CM5, or Pi camera, or external GPU, or really slow internet.
2. Helping add new features that QEMU offers (serial/bluetooth passthrough, more virtio drivers, etc.) in Pull Requests
3. Supporting new users and spreading the word

Botspot, 2025-02-26:
@gitbls Were you able to try the `boot-nodisplay` with the `connect` mode, as explained in the readme, or for now have you remained using the `boot` mode? I'd like to have your thoughts on how the performance compares. RDP seems faster mostr of the time but it might be placebo.

gitbls, 2025-02-26:
> You encountered 2 issues:
> 
> 1. Some kind of issue with the Windows Defender removal. Fixed in [d7b054c](https://github.com/Botspot/bvm/commit/d7b054c2239b98b2467192483e89d5635e72ea64)
> 2. One-time issue with how the script restarts itself after receiving an update. Already fixed in commit [d2b61e1](https://github.com/Botspot/bvm/commit/d2b61e11c9eb278f2162639875b54b746fa220eb), but anyone who cloned the repo before that will encounter this one single time.
> 
> How to help? I can think of a few ways:
> 
> 1. Testing on system setups that I don't have. For example, a CM5, or Pi camera, or external GPU, or really slow internet.
> 2. Helping add new features that QEMU offers (serial/bluetooth passthrough, more virtio drivers, etc.) in Pull Requests
> 3. Supporting new users and spreading the word

OK, I'll re-clone after addressing your next question (connect mode) and run through it again.

Unfortunately I don't have any of those hw bits, and I'll be mostly out of pocket until mid-March (which I conveniently forgot about!).

Botspot, 2025-02-26:
There should never be a reason to re-clone, unless auto-update fails which I do not think it is.

gitbls, 2025-02-26:
> [@gitbls](https://github.com/gitbls) Were you able to try the `boot-nodisplay` with the `connect` mode, as explained in the readme, or for now have you remained using the `boot` mode? I'd like to have your thoughts on how the performance compares. RDP seems faster most of the time, but it might be placebo.

Just tried `boot-nodisplay` and `connect`. It worked much better once I set the add_freerdp_flags=(+dynamic-resolution). Without that the wlfreerdp client croaked when I changed the window size, but now it's pretty good (and larger...thanks!)

Will everything I've done work with X11 rather than Wayland? 

Also, is it safe to run Windows Update in the VM? Nothing urgent there, just curious if/how it works.

Botspot, 2025-02-27:
`dynamic-resolution` should already enabled. Huh. Please explain what is happening in more detail.

Yes, X11 should work, but I have not tried it. Be aware BVM installs xfreerdp instead of wlfreerdp, so if you switch back and forth there will be some APT activity.

Yes, updates seem to work fine for me. Also I have not had any problems doing unsafe shutdowns.

gitbls, 2025-02-27:
> `dynamic-resolution` should already enabled. Huh. Please explain what is happening in more detail.
>
> Yes, X11 should work, but I have not tried it. Be aware BVM installs xfreerdp instead of [...]

OK, backing out my change to bvm-config (dynamic resolution was the only thing I set) and dynamic resolution change is working now. Not sure why I thought I had to enable it. 🫤🤔

gitbls, 2025-02-27:
Confirmed that switching to X11 works as expected.

Botspot, 2025-02-27:
Comment back here if you have any additional incidental findings, or open a new issue if you run into any further problems.
I'm particularly interested to find out all the ways people actually put this to use!

closed 2025-02-27 by Botspot