Botspot/bvm #4: ./ in README.md is unnecessary

thagrol, 2025-02-27:
In several places in README.md you have `./bvm/bvm`. The leading `./` is unnecessary. `bvm/bvm` has the same effect.

In general, a `./` prefix is only required when the thing you're running is in the CWD and has the same filename as a command in a directory in $PATH and/or `.` is not already at the start of $PATH.

A minor point, I know.

Botspot, 2025-02-27:
Thank you for reaching out.
My reason for having this was because the `bvm` script needs to find its residing folder using the `$0` variable. In the distant past, with another project, I had issues when running it with relative paths, and I suppose this misconception stuck with me. Just tested it with `bvm/bvm` and while it does change the value of `$0`, the script seems to still derive its containing directory correctly.

Will change shortly, after a few more tests.

closed 2025-02-27 by Botspot

Botspot, 2025-02-27:
Yes, I have used $PWD plenty in the past.
In this case `readlink -f "$(dirname "$0")"` does a splendid job of getting full paths, and it does already rely on $PWD as needed.