Botspot/bvm #5: Firstboot Stuck on Raspberry Pi 4  &  Pi 400

geev-gv, 2025-02-27:
`
ubu@raspberrypi:~ $ ./bvm/bvm download ~/win11
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██                                                      
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███                                                      
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█                                                      
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█                                                      
 █████🭠   🭖██🭡   █▉     🮋█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"Achievement unlocked: Running Windows where it absolutely should not be."
BVM mode download
Downloading Windows 11 ARM64 (English (United States))
  - Parsing download page: https://www.microsoft.com/en-us/software-download/windows11arm64
  - Getting Product edition ID: 3131
  - Permit Session ID: 00100cb3-5e05-44a2-b2d0-5d434e93ae2a
  - Getting language SKU ID: 18624
  - Getting ISO download link...
  - URL: https://software.download.prss.microsoft.com/dbazure/Win11_24H2_English_Arm64.iso
/home/ubu/win11/ins 100%[===================>]   5.08G  3.56MB/s    in 30m 45s 
2025-02-26 15:28:48 URL:https://software.download.prss.microsoft.com/dbazure/Win11_24H2_English_Arm64.iso?t=6a4f416f-0744-4983-b123-1d25f5181e1e&P1=1740668283&P2=601&P3=2&P4=vKojGG3qTHiMQZ1C6GhzVv2i4XR0MpQYWKBacd1dzhU0T9MuLE%2fsBUsrif4uKV5xFFqoF63Ys3lvZJQc5hApoiOmsYKzf5joi%2bykmbFiUNu%2bT9s0Wghq2aJkvMR1Ht3KyGnkrrukGlaRp5lPi%2f%2f4IKXzJLk2THmlS8TNQFH0DhphH5fX4M%2btPZO4tTMFfSrcDtsu9fWAoHLzArJCgJA135aEeTeyV093EG6e6%2f0a0HFH4oeXJjqy7nvxK4tSqOBilciA6ZGi%2bENbdrBrACaKLf1lt%2bvkHywC%2fw106hosQcpuDmZ%2bI9wUGE0pY%2buhq3HdPUFagn90%2faOjKGGZLZo%2f3g%3d%3d [5460387840/5460387840] -> "/home/ubu/win11/installer.iso" [1]
Downloading virtio drivers
/home/ubu/win11/virtio-win.iso                              100%[=========================================================================================================================================>] 690.88M   369KB/s    in 33m 25s 
2025-02-26 16:02:14 URL:https://fedorapeople.org/groups/virt/virtio-win/direct-downloads/archive-virtio/virtio-win-0.1.266-1/virtio-win-0.1.266.iso [724434944/724434944] -> "/home/ubu/win11/virtio-win.iso" [1]
Downloading Windows debloating script
Cloning into 'Win11Debloat'...
remote: Enumerating objects: 893, done.
remote: Counting objects: 100% (244/244), done.
remote: Compressing objects: 100% (73/73), done.
remote: Total 893 (delta 202), reused 172 (delta 171), pack-reused 649 (from 1)
Receiving objects: 100% (893/893), 264.79 KiB | 35.00 KiB/s, done.
Resolving deltas: 100% (643/643), done.
Extracting VirtIO drivers...
  - Balloon
  - NetKVM
  - pvpanic
  - viogpudo
  - vioinput
  - viomem
  - viorng
  - vioscsi
  - vioserial
  - viostor
You should now be ready for the next step: ./bvm/bvm prepare /home/ubu/win11
ubu@raspberrypi:~ $ ./bvm/bvm prepare /home/ubu/win11/
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██                                                      
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███                                                      
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█                                                      
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█                                                      
 █████🭠   🭖██🭡   █▉     🮋█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"Because there is no true difference between ‘unsupported’ and ‘fun challenge.’"
BVM mode prepare
Modifying Windows ISO to boot to installer without keypress:
  - Searching for efisys_noprompt.bin to extract...
  - negative match at byte 5422041088
  - positive match at byte 5424738304
  - Extracting it...
  - Checksums match! Now searching ISO for sections to replace...
  - positive match at byte 1087488, replacing the next 1720320 bytes...
  - negative match at byte 5421072384, continuing to search...
  - positive match at byte 5423017984, replacing the next 1720320 bytes...
Done
Making unattended.iso... Warning: creating filesystem that does not conform to ISO-9660.
Done
Setting up main hard drive disk.qcow2
Allocating 40GB for main install drive... Done
You should now be ready for the next step: ./bvm/bvm firstboot /home/ubu/win11/
ubu@raspberrypi:~ $ 

`

I've encountered an issue with the first boot process on my Raspberry Pi 4 -8 & p-400 . Previously, when upgraded from w10 to w11 in 2022 , the version was # 26100.2033_en-us and the activated version was 'Windows Workstation' & the maximum upgrade was 22H2.
Guess the problem arises from trying 24H2  on rPi4 .(?)

Botspot, 2025-02-27:
> Guess the problem arises from trying 24H2 on rPi4 .(?)

Yes, based on your findings lining up with mine, it would appear so. To get the latest ISO, Microsoft has a convenient download now, but for an older ISO I'm not aware of anything more reliable than uupdump, which was a nightmare back when I used it in WoR-Flasher.

For now for you, if you have a bit of time and want to help, please try going to https://uupdump.net/, download the version of Windows 11 that you think will work, and replace the `/home/ubu/win11/installer.iso` file with the file you downloaded, then see if that works.

geev-gv, 2025-02-27:
`   38  history
ubu@raspberrypi:~ $ ./bvm/bvm prepare /home/ubu/woa
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██                                                      
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███                                                      
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█                                                      
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█                                                      
 █████🭠   🭖██🭡   █▉     🮋█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"Think of this as a very elaborate benchmark... for both you and your hardware."
BVM mode prepare
Modifying Windows ISO to boot to installer without keypress:
  - Searching for efisys_noprompt.bin to extract...
  - negative match at byte 5020139928
  - negative match at byte 5022787992
Failed to extract efisys_noprompt.bin from ISO!
ubu@raspberrypi:~ $ 

`
No joy with the old ISO too.

Botspot, 2025-02-27:
Hmm, that's just my fancy function for making the installer not require a keypress on boot. It is not crucial for booting.
Please comment out the `patch_iso_noprompt "$vmdir/installer.iso" || exit 1` line in the script and try again.

Botspot, 2025-02-27:
I'm working on a way to download an older Microsoft ESD image, convert to ISO, and use that for Pi4 and below.

geev-gv, 2025-02-27:
After commenting out your 'fancy function' ,as suggested, everything went on smoothly and I on the first GUI boot,  thank you.
Please include 22000.51_arm64_en-US_pro also in the list of older Microsoft ESD images as it has been working on rPi4 /400.

Botspot, 2025-02-27:
Commit https://github.com/Botspot/bvm/commit/d0b0a1ff228bbebfc2250b97255048032f3df3c7 should fix this and download version Windows 22631.2861 for devices that lack A76 CPU cores.
@geev-gv You are welcome to redo the download step and make sure this slightly newer windows version works on your end, or you could stay on 22000 and it should work fine going forward.

One thing I am interested to know is if Windows eventually tries to upgrade people from an old build to a newer one that's incompatible with the CPU. If so I would need to find a registry hack to disable updates.

geev-gv, 2025-02-28:
`
ubu@raspberrypi:~ $ bvm/bvm prepare /home/ubu/win11
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█
 █████🭠   🭖██🭡   █▉     🮋█
BOTSPOT  VIRTUAL  MACHINE
tee: /dev/stderr: Permission denied
BVM mode prepare
Making unattended.iso... Warning: creating filesystem that does not conform to ISO-9660.
Done
Setting up main hard drive disk.qcow2
Allocating 40GB for main install drive... Done
You should now be ready for the next step: bvm/bvm firstboot /home/ubu/win11
ubu@raspberrypi:~ $ bvm/bvm firstboot /home/ubu/win11
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█
 █████🭠   🭖██🭡   █▉     🮋█
BOTSPOT  VIRTUAL  MACHINE
tee: /dev/stderr: Permission denied
BVM mode firstboot
Windows should install 100% automatically. This will take several hours.
Authorization required, but no authorization protocol specified

qemu-system-aarch64: OpenGL is not supported by display backend 'gtk'
QEMU did not exit successfully.
ubu@raspberrypi:~ $ sudo bvm/bvm firstboot /home/ubu/win11
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█
 █████🭠   🭖██🭡   █▉     🮋█
BOTSPOT  VIRTUAL  MACHINE
"The only thing more surprising than this working... is how well it works."
bvm/bvm: line 535: /root/.local/share/applications/bvm.desktop: No such file or directory
ubu@raspberrypi:~ $

`
This seems to be a new one that came up while trying the fix on a rPi-400.

Botspot, 2025-02-28:
It seems like qemu on your system is missing support for running in a openGL window. I'm curious what version of QEMU it is. @geev-gv please run this to find out:
```
qemu-system-aarch64 --version
```

geev-gv, 2025-03-01:
`
gv@raspberrypi:~ $ !54
qemu-system-aarch64 --version
QEMU emulator version 9.2.0 (Debian 1:9.2.0+ds-3~bpo12+2)
Copyright (c) 2003-2024 Fabrice Bellard and the QEMU Project developers
gv@raspberrypi:~ $ !63
~/.local/share/bvm/bvm boot-nodisplay ~/win11
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█
 █████🭠   🭖██🭡   █▉     🮋█
BOTSPOT  VIRTUAL  MACHINE
"Let’s find out if your hardware has quality memory and cooling. >:)"
BVM mode boot-nodisplay
qemu-system-aarch64: -netdev user,id=nic,hostfwd=tcp:127.0.0.1:3389-:3389: Could not set up host forwarding rule 'tcp:127.0.0.1:3389-:3389'
QEMU did not exit successfully.
gv@raspberrypi:~ $
`

The previous install ,on another ssd, with my old iso works and hope this , https://h5p.org/node/1521514 , is of some use.

Botspot, 2025-03-01:
@geev-gv, the new error we are seeing is:
```
qemu-system-aarch64: -netdev user,id=nic,hostfwd=tcp:127.0.0.1:3389-:3389: Could not set up host forwarding rule 'tcp:127.0.0.1:3389-:3389'
```
Which seems a bit clear. I think QEMU could not route its internet traffic through port 3389 on this system, perhaps because some other app is doing so.

Reading and changing this part of the config file may solve the problem.
```
#If you have mutliple VMs, give each one a unique value.
rdp_port=3389
#Make sure this value does not conflict with other network services shown when running netstat -tuln
```

- Me personally, I would first check if any services are using port 3389 already. To do that run `netstat -tuln | grep :3389`
- Then as a second step, try changing the value of rdp_port in the config file and see if it makes a difference.
- If you keep getting this error, and there is no conflicting service, then maybe this is a permissions issue or firewall issue on your system.

Botspot, 2025-03-02:
You're on Ubuntu. Maybe security is getting in the way, like AppArmor or firewall. I have no experience with that. If my suggestions above don't work, unless you need to stay on Ubuntu maybe consider switching to Pi OS, as that is known to run BVM just fine.

madcock, 2025-03-02:
I also got the "qemu-system-aarch64: OpenGL is not supported by display backend 'gtk'" message when running firstboot. I'm on a CM4 with 8GB RAM, and my QEMU version matches the one mentioned above:
"QEMU emulator version 9.2.0 (Debian 1:9.2.0+ds-3~bpo12+2)"

Maybe I missed something in the thread, but how did you get past that error? I haven't tried boot-nodisplay since it didn't successfully complete the firstboot.

Let me know if there's any other useful information I could provide.

Thanks!

Botspot, 2025-03-02:
Could one of you try `sudo apt install -t bookworm-backports qemu-system-modules-opengl`

Botspot, 2025-03-02:
> So it might be worth mentioning in the usage instructions that the firstboot command should be run in the windowing environment on the device, rather than remotely via SSH.

It takes idiots to make something idiot-proof. :) 
I'll make the script warn is no display is detected. Good idea.

madcock, 2025-03-02:
I updated my original post, but the issue for me was because I was SSHing in, and not running the command directly on the device with the display up. I'll report back if there are issues once it's done setting up, but for now it's installing Windows and I guess I'll know what the result is in a few hours. :)

Botspot, 2025-03-03:
@geev-gv please open a new issue if you try my suggestions and still have problems getting QEMU running.

commit d0b0a1ff (2025-02-27): Support Pi4 and other non-A76 ARM cores with old windows build

Fixes honoring disable_updates value
Adds non-A76 core detection, and downloading of windows 11 version 22631
if detected
Adds wimtools dependency
Ensure vmdir is not FAT partition
Try to support Rockchip RK3399 by only running on performance cores
(untested)
Closes #5
closed 2025-03-03 by Botspot