Botspot/bvm #10: Progress report (mainly positive) plus Search and usb msd queries

DeeJay, 2025-02-28:
First: Thank you for your efforts. To a large extent it 'Just Works' as documented. I would not have had the determination time or background knowledge to get this far. I am 'up and running' with a Win11ARM Desktop.

I am running the headless + connect option.

Some comments. On an 8GB RPi[5] I have peak usage of 90% of the RasPiOS default 512M swap allocation. I therefore underscore your comment on this topic under Tips in the README.

Possible issue for investigation. Sometimes when a Search bar is open, in Edge or the Windows GUI Home menu, it spontaneously fills with 'quote' characters. Hitting Esc seems to stop this, and the garbage characters can then be deleted.

Use of USB removable Mass Storage. I plugged in a thumbdrive/memory stick. It shows up in the filesystem of the RasPiOS host at /dev/sdb1 mounted at /media/david/EB61-9206.  On the Win11ARM instance it shows up under This PC as 'EB61-9206' and I can view a directory listing. Good so far...    I have installed the Windows version of Raspberry Pi Imager, v1.8.5. Imager does not find any eligible mass storage to flash an image onto. Should this be possible?

Botspot, 2025-02-28:
> Some comments. On an 8GB RPi[5] I have peak usage of 90% of the RasPiOS default 512M swap allocation. I therefore underscore your comment on this topic under Tips in the README.

RAM usage just got a lot better after this change I just published. https://github.com/Botspot/bvm/commit/a4c85011e769269ef3655c8cb167cba979f93b82 Just run the script again and you should now see qemu not need to dip into swap at all.

> Possible issue for investigation. Sometimes when a Search bar is open, in Edge or the Windows GUI Home menu, it spontaneously fills with 'quote' characters. Hitting Esc seems to stop this, and the garbage characters can then be deleted.

I have not seen that happen. My guess is it could be an issue with the RDP viewer. Try seeing if it ever occurs with the `connect-remmina` mode.

> Use of USB removable Mass Storage. I plugged in a thumbdrive/memory stick. It shows up in the filesystem of the RasPiOS host at /dev/sdb1 mounted at /media/david/EB61-9206. On the Win11ARM instance it shows up under This PC as 'EB61-9206' and I can view a directory listing. Good so far...

that is accomplished through the RDP viewer sharing files to windows using a network share.

> I have installed the Windows version of Raspberry Pi Imager, v1.8.5. Imager does not find any eligible mass storage to flash an image onto. Should this be possible?

This requires me to finish implementing USB passthrough.

Botspot, 2025-03-01:
USB passthrough should now be implemented and it looks like it works on Imager.

[image]

closed 2025-03-02 by Botspot