Botspot/bvm #11: Bluetooth

DeeJay, 2025-02-28:
Bluetooth.

The Win11Arm instance System Settings say Bluetooth is not supported. Should it be? (The RasPiOS host is an RPi[5]-8, so the underlying hardware can do it.)

(Would be good for getting Audio out of the system.)

Botspot, 2025-02-28:
I knew it! I knew someone would want to do bluetooth passthrough. It should be possible but I am not sure how at the moment. I actually expected someone to ask for this and left a note in the README asking for QEMU experts to step forward. We'll see if anyone does.

OxIceBear, 2025-03-03:
I have bluetooth devices... A ton of them, i can help you with it if you want to.

RaginCajunSteve, 2025-03-09:
Also, Link to Windows Phone App which is required for tons of authentication requires the use of bluetooth pass through to make the reason for having windows on raspberry pi required for Military use case.

krishnanshyadav, 2025-06-19:
i wanted to connect my ev3 with rpi4 in win11 with a app can you help me with it