Botspot/bvm #13: firstboot:  "The computer restarted unexpectedly or encountered an unexpected error. Windows installation cannot proceed"

tinker2much, 2025-02-28:
An error occurred some long time into firstboot. I wasn't watching it so I don't know exactly how far it had gotten. Here's a screenshot:

[image]
I clicked OK, it seemed to reboot and start the install again. I went to bed. It was at the same place this morning. When I move the message box,  looking for something informative, there's nothing behind it.

Botspot, 2025-02-28:
It works fine on my Pi 4. Try deleting/renaming your vm download folder and trying again, to see if it could be internet-related.

tinker2much, 2025-02-28:
> Are you using Pi 4 or Pi 5? The reason I ask is they now each use completely different sources for the installer ISO.
> 
> Edit: looks like you said Pi 4 8GB. I will need to investigate.

I misspoke. It's a pi4-4GB.

tinker2much, 2025-02-28:
When I did a Ctrl-C on bvm, further information appeared:

> "Think of this as a very elaborate benchmark... for both you and your hardware."
> BVM mode firstboot
> Windows should install 100% automatically. This will take several hours.
> ^Cqemu: terminating on signal 2
> QEMU closed.
> The disk contains an unclean file system (0, 0).
> Metadata kept in Windows cache, refused to mount.
> Falling back to read-only mount because the NTFS partition is in an
> unsafe state. Please resume and shutdown Windows fully (no hibernation
> or fast restarting.)
> Could not mount read-write, trying read-only
> mount: (hint) your fstab has been modified, but systemd still uses
>        the old version; use 'systemctl daemon-reload' to reload.
> It appears that Windows installation was not completed in disk.qcow2. C:/Program Files/Qemu-ga/qemu-ga.exe does not exist.
> 
> real	777m19.846s
> user	478m57.582s
> sys	89m46.446s
> 
>

tinker2much, 2025-02-28:
> It works fine on my Pi 4. Try deleting/renaming your vm download folder and trying again, to see if it could be internet-related.

I did so, followed through all the steps, and everything worked fine this time. I'm now exploring Windows 11 Arm on my pi4-4GB. 

Completely awesome! Thank you for this amazing effort. And thank you for your patient support.

Botspot, 2025-02-28:
Okay it either could have been a corrupted download the first time, or working from one of the fixes I made since first time. Not sure.

closed 2025-02-28 by Botspot