Botspot/bvm #14: First time usage notes (on a fresh Pi 5 install)

geerlingguy, 2025-02-28:
A few notes:

  - I had to manually run `sudo apt update` after the first `bvm/bvm` attempt, as my package index was not up to date. Could that be added to the initial setup process?
  - Perhaps a step of either moving or linking `bvm` into `/usr/bin/bvm` or something like that, so you don't have to remember to enter the path to `bvm` (`bvm/bvm` or `./bvm`)?
  - Have you tested an install on microSD versus NVMe? Does it make a huge difference?

geerlingguy, 2025-02-28:
Blog post: [Windows 11 Arm VMs on a Raspberry Pi, with BVM](https://www.jeffgeerling.com/blog/2025/windows-11-arm-vms-on-raspberry-pi-bvm)

Video: https://www.youtube.com/watch?v=mkfILjKJ8nc

Botspot, 2025-02-28:
> A few notes:
> 
> * I had to manually run `sudo apt update` after the first `bvm/bvm` attempt, as my package index was not up to date. Could that be added to the initial setup process?

Will fix. I watched your video. Also just so you know, you could maximize the RDP window if you wanted. :)
Minecraft mouse grabbing should work but it looks like you can't use RDP for it. Try booting the VM directly - with the `boot-gtk` mode instead of the `boot-nodisplay` and `connect` modes. That should fix minecraft at least.

> * Perhaps a step of either moving or linking `bvm` into `/usr/bin/bvm` or something like that, so you don't have to remember to enter the path to `bvm` (`bvm/bvm` or `./bvm`)?

The script-apps I write generally stay in ~/ so that updates with `git pull` don't need root, and so the user can more easily find and edit the scripts. Also wayland applications cannot run as root. I could however symlink bvm to ~/.local/bin, but unless I'm mistaken that would require a reboot for the directory to be added to $PATH, if the directory started empty.

Great video. Also thanks for addressing my curiosity on if external GPUs could work. Are you aware of any GPUs that have a Windows ARM driver? (besides snapdragon's integrated graphics)
In any case, just making sure you saw that virtualized graphics are a possibility and I think all we need is someone capable of compiling it. Then the VM may be able to get graphics accel by communicating with Vulkan.

geerlingguy, 2025-02-28:
> you could maximize the RDP window if you wanted. :)

Haha but where's the fun in actually seeing anything you're doing???

> Are you aware of any GPUs at all that have a Windows ARM driver?

So far the _only_ thing besides Snapdragon drivers is a driver for the ASPEED integrated 2D graphics chip on some server motherboards... haven't seen anything at all for other Arm SoCs or any standalone GPUs. 

I *know* that a driver for at least high end Nvidia cards exists for Windows on Arm—it's in use on Azure. But it has not been released to the public ever :(

Botspot, 2025-03-01:
USB passthrough is now supported. Currently it needs to be enabled per device in the config file. I might make a GUI setting for this.
https://github.com/Botspot/bvm/commit/9389caa5bcc7548363b0fad999f0e32591f16487
Tested and working on a flash drive and USB sound card.

farjanatech, 2025-03-01:
Here is the GPU Driver for Windows 11 ARM I found from the following ISO file 
(Install the virtio-gpu driver by going to the \viogpudo\w11\ARM64 folder inside ISO)

Source: Download the latest "virtio-win.iso" from here:
https://fedorapeople.org/groups/virt/virtio-win/direct-downloads/latest-virtio

farjanatech, 2025-03-01:
Here is the Proof that I am using the above-mentioned GPU Driver with Windows 11 QEMU in my Raspberry Pi 5 (8GB) board.
https://prnt.sc/4RSrKaFEMJwN

Botspot, 2025-03-01:
> Here is the Proof that I am using the above-mentioned GPU Driver with Windows 11 QEMU in my Raspberry Pi 5 (8GB) board. https://prnt.sc/4RSrKaFEMJwN

@farjanatech, Sorry to rain on your birthday party, but that is a display adapter, and is not the same as graphics acceleration. BVM already installs virtio-gpu by default, in order for the `boot-gtk` mode to work. If you go to a web browser and type `about:gpu` into the address bar, you would see that nothing is hardware accelerated.

Botspot, 2025-03-02:
Since the last comment, I have:
- Added a `bvm` symlink to ~/.local/bin. This won't be the primary recommended way to run it, in case that directory is not on $PATH, but it should come in handy for many.
- Fixed the `connect` mode to detect VM reboots and relaunch the RDP viewer when ready. This makes the GUI a lot more usable as the primary usage method. So far, more people have preferred the GUI than I thought. I think my mind was a bit warped by the RPi Forums into thinking everyone prefers CLI, haha.
- Made sure `sudo apt update` is run first, but only if needed.

I think that covers everything. Looking forward to seeing how the community responds to this. I think it has nearly unlimited potential, as this now squarely beats Windows On Raspberry in terms of usefulness and utility. Most folks just don't realize that yet.

Botspot, 2025-03-02:
Also added RK3588 support. I hope someone prominent in that community takes notice. It could help them out a lot.

closed 2025-03-02 by Botspot

farjanatech, 2025-03-02:
Honestly, you are the best man who cares about the Windows on Raspberry community.  Because you are often updating BVM so I decided to try this after 3 days. To make sure I can utilize last fix or option you add.

Botspot, 2025-03-02:
> Honestly, you are the best man who cares about the Windows on Raspberry community. Because you are often updating BVM so I decided to try this after 3 days. To make sure I can utilize last fix or option you add.

It is ironic that I made the WoR-Flasher. It means that now two of my projects are competing with each other somewhat. It would be really funny if I advertised BVM in WoR-Flasher. Might need to consider that honestly.

But @farjanatech, I didn't mean the WoR community, I meant the RK3588 community - these are non-raspberry pis and BVM should now run on them. What I meant was that I hope those users somehow find out about it and benefit from it too.

Botspot, 2025-03-02:
Here we go.

[image]

geerlingguy, 2025-03-02:
> this now squarely beats Windows On Raspberry in terms of usefulness and utility. Most folks just don't realize that yet.

Indeed. I have many more use cases that can be solved by "and you can run that one little Windows utility in a VM when you need it" than "I would like to run full Windows 11 Pro directly on a Raspberry Pi". The latter is fun, to be sure, but the former is more useful and solves use cases that some people have said hampered their ability to deploy a Pi somewhere.