Botspot/bvm #15: Error installing from Pi-Apps and Will not start from git clone command

daverichathome, 2025-02-28:
Am getting the following error when trying to install any application from "pi-apps"

Unable to open a terminal.
Debug output below.
Default x-terminal-emulator: gnome-terminal.wrapper
Using gnome-terminal.wrapper
[91mterminal-run: No terminal detected as it never created the pid file within 10 seconds.[39m
# Error constructing proxy for org.gnome.Terminal:/org/gnome/Terminal/Factory0: Error calling StartServiceByName for org.gnome.Terminal: Timeout was reached

Ran git clone command and it installed, but when I start BVM it just hangs on the Start screen and will not proceed.

Thanks

Botspot, 2025-03-01:
A few people have been reporting this. https://github.com/Botspot/pi-apps/issues/2717 I can't get it to happen on one of my Pies though, and I think it's related to the exact PiOS image you are using.

Please run this and send the output back to me.
```
cat /etc/os-release /etc/rpi-issue
```

daverichathome, 2025-03-01:
cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

cat /etc/rpi-issue 
Raspberry Pi reference 2024-07-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 48efb5fc5485fafdc9de8ad481eb5c09e1182656, stage4

Botspot, 2025-03-01:
Okay I've downloaded that exact Pi OS image, except the difference was that gnome-terminal was nowhere to be found, and the default terminal was lxterminal. Very strange indeed.

@daverichathome let's try to figure out what is responsible for installing this, because it seems like gnome-terminal did not come pre-installed. (unless they edited the disk image)
Please run these commands and send back the output here.
```
grep "gnome-terminal" -B5 -A5 /var/log/apt/history.log
zgrep "gnome-terminal" -B5 -A5 /var/log/apt/history.log.*gz
```

daverichathome, 2025-03-01:
/var/log/apt/history.log.5.gz-Install: linux-headers-6.6.51+rpt-rpi-v8:arm64 (1:6.6.51-1+rpt3, automatic), linux-image-6.6.51+rpt-rpi-2712:arm64 (1:6.6.51-1+rpt3, automatic), python3-webcolors:arm64 (1.11.1-1, automatic), libfuse2:arm64 (2.9.9-6+b1, automatic), libspa-0.2-libcamera:arm64 (1.2.4-1~bpo12+1+rpt1, automatic), libconfig++9v5:arm64 (1.5-0.4, automatic), wfplug-squeek:arm64 (0.5, automatic), python3-attr:arm64 (22.2.0-1, automatic), python3-jsonschema:arm64 (4.10.3-1, automatic), linux-kbuild-6.6.51+rpt:arm64 (1:6.6.51-1+rpt3, automatic), python3-json-pointer:arm64 (2.3-2, automatic), linux-headers-6.6.51+rpt-rpi-2712:arm64 (1:6.6.51-1+rpt3, automatic), python3-uritemplate:arm64 (4.1.1-2, automatic), python3-libarchive-c:arm64 (2.9-0.1, automatic), linux-image-6.6.51+rpt-rpi-v8:arm64 (1:6.6.51-1+rpt3, automatic), libxml++2.6-2v5:arm64 (2.40.1-3, automatic), wfplug-connect:arm64 (0.9, automatic), python3-rfc3987:arm64 (1.3.8-2, automatic), python3-pyrsistent:arm64 (0.18.1-1+b3, automatic), python3-tqdm:arm64 (4.64.1-1, automatic), linux-headers-6.6.51+rpt-common-rpi:arm64 (1:6.6.51-1+rpt3, automatic), libffado2:arm64 (2.4.7-1, automatic)
/var/log/apt/history.log.5.gz-Upgrade: libcups2:arm64 (2.4.2-3+deb12u7, 2.4.2-3+deb12u8), docker-compose-plugin:arm64 (2.29.2-1~debian.12~bookworm, 2.29.7-1~debian.12~bookworm), cups-filters:arm64 (1.28.17-3, 1.28.17-3+deb12u1), labwc:arm64 (0.7.2-1+rpt4, 0.7.2-1+rpt7), libneatvnc0:arm64 (0.9.0+dfsg-1~rpt2, 0.9.0+dfsg-1~rpt3), docker-ce-cli:arm64 (5:27.2.1-1~debian.12~bookworm, 5:27.3.1-1~debian.12~bookworm), chromium-sandbox:arm64 (1:128.0.6613.137-1~deb12u1+rpt2, 1:130.0.6723.58-1~deb12u1+rpt2), rpi-eeprom:arm64 (25.2-1, 26.1-1), cups-common:arm64 (2.4.2-3+deb12u7, 2.4.2-3+deb12u8), gui-pkinst:arm64 (0.17, 0.18), userconf-pi:arm64 (0.9, 0.10), libpipewire-0.3-common:arm64 (0.3.65-3+rpt8+deb12u1, 1.2.4-1~bpo12+1+rpt1), pipewire-pulse:arm64 (0.3.65-3+rpt8+deb12u1, 1.2.4-1~bpo12+1+rpt1), pipanel:arm64 (1:1.48, 1:1.49), libwayland-cursor0:arm64 (1.22.0-2.1~bpo12+rpt1, 1.23.0-1~bpo12+1+rpt1), libwlroots12:arm64 (0.17.3-1~bpo12+rpt5, 0.17.4-1+rpt1), libheif1:arm64 (1.15.1-1, 1.15.1-1+deb12u1), raspi-config:arm64 (20240708, 20241025), libexpat1:arm64 (2.5.0-1, 2.5.0-1+deb12u1), linux-headers-rpi-v8:arm64 (1:6.6.47-1+rpt1, 1:6.6.51-1+rpt3), libfm-gtk-data:arm64 (1.3.2-1+rpt13, 1.3.2-1+rpt16), cups-client:arm64 (2.4.2-3+deb12u7, 2.4.2-3+deb12u8), cups-ppdc:arm64 (2.4.2-3+deb12u7, 2.4.2-3+deb12u8), libwayland-server0:arm64 (1.22.0-2.1~bpo12+rpt1, 1.23.0-1~bpo12+1+rpt1), python3-reportlab-accel:arm64 (3.6.12-1+b1, 3.6.12-1+deb12u1), cups-daemon:arm64 (2.4.2-3+deb12u7, 2.4.2-3+deb12u8), rc-gui:arm64 (1.85, 1.89), libjavascriptcoregtk-4.1-0:arm64 (2.44.3-1~deb12u1, 2.46.0-2~deb12u1), libdrm-nouveau2:arm64 (2.4.114-1+b1, 2.4.123-1~bpo12+1+rpt1), python3-reportlab:arm64 (3.6.12-1, 3.6.12-1+deb12u1), libfontembed1:arm64 (1.28.17-3, 1.28.17-3+deb12u1), raspberrypi-ui-mods:arm64 (1.20240808, 1.20241028+1), pipewire:arm64 (0.3.65-3+rpt8+deb12u1, 1.2.4-1~bpo12+1+rpt1), wf-panel-pi:arm64 (0.76, 0.87), rpi-connect:arm64 (1.3.2, 2.0.0), linux-headers-rpi-2712:arm64 (1:6.6.47-1+rpt1, 1:6.6.51-1+rpt3), libcamera-ipa:arm64 (0.3.1+rpt20240906-1, 0.3.2+rpt20240927-1), pi-greeter:arm64 (0.23, 0.24), linux-image-rpi-2712:arm64 (1:6.6.47-1+rpt1, 1:6.6.51-1+rpt3), python3-picamera2:arm64 (0.3.21-1, 0.3.22-2), chromium:arm64 (1:128.0.6613.137-1~deb12u1+rpt2, 1:130.0.6723.58-1~deb12u1+rpt2), rpd-plym-splash:arm64 (0.39, 0.41), cups-filters-core-drivers:arm64 (1.28.17-3, 1.28.17-3+deb12u1), docker-buildx-plugin:arm64 (0.16.2-1~debian.12~bookworm, 0.17.1-1~debian.12~bookworm), cups-ipp-utils:arm64 (2.4.2-3+deb12u7, 2.4.2-3+deb12u8), wayvnc:arm64 (0.9.0~dev-1, 0.9.0~dev-1+rpt1), docker-ce:arm64 (5:27.2.1-1~debian.12~bookworm, 5:27.3.1-1~debian.12~bookworm), libcamera0.3:arm64 (0.3.1+rpt20240906-1, 0.3.2+rpt20240927-1), firefox:arm64 (130.0-1+rpt7, 131.0.3-1+rpt7), libdrm-common:arm64 (2.4.114-1, 2.4.123-1~bpo12+1+rpt1), pipewire-libcamera:arm64 (0.3.65-3+rpt8+deb12u1, 1.2.4-1~bpo12+1+rpt1), lxinput:arm64 (0.3.5-1+rpt22, 0.3.5-1+rpt24), libcamera-tools:arm64 (0.3.1+rpt20240906-1, 0.3.2+rpt20240927-1), libspa-0.2-bluetooth:arm64 (0.3.65-3+rpt8+deb12u1, 1.2.4-1~bpo12+1+rpt1), rp-prefapps:arm64 (0.59, 0.64), agnostics:arm64 (0.16, 0.17), docker-ce-rootless-extras:arm64 (5:27.2.1-1~debian.12~bookworm, 5:27.3.1-1~debian.12~bookworm), rp-bookshelf:arm64 (0.23, 0.24), raspi-firmware:arm64 (1:1.20240902-1, 1:1.20240924-1), libfm-gtk4:arm64 (1.3.2-1+rpt13, 1.3.2-1+rpt16), python3-renderpm:arm64 (3.6.12-1+b1, 3.6.12-1+deb12u1), cups-browsed:arm64 (1.28.17-3, 1.28.17-3+deb12u1), pipewire-bin:arm64 (0.3.65-3+rpt8+deb12u1, 1.2.4-1~bpo12+1+rpt1), lightdm:arm64 (1.26.0-8+rpt3, 1.26.0-8+rpt5), libcupsimage2:arm64 (2.4.2-3+deb12u7, 2.4.2-3+deb12u8), libwebkit2gtk-4.1-0:arm64 (2.44.3-1~deb12u1, 2.46.0-2~deb12u1), pishutdown:arm64 (0.29, 0.30), chromium-l10n:arm64 (1:128.0.6613.137-1~deb12u1+rpt2, 1:130.0.6723.58-1~deb12u1+rpt2), linux-image-rpi-v8:arm64 (1:6.6.47-1+rpt1, 1:6.6.51-1+rpt3), libwayland-egl1:arm64 (1.22.0-2.1~bpo12+rpt1, 1.23.0-1~bpo12+1+rpt1), chromium-common:arm64 (1:128.0.6613.137-1~deb12u1+rpt2, 1:130.0.6723.58-1~deb12u1+rpt2), piwiz:arm64 (0.75, 0.84), libspa-0.2-modules:arm64 (0.3.65-3+rpt8+deb12u1, 1.2.4-1~bpo12+1+rpt1), libfm-extra4:arm64 (1.3.2-1+rpt13, 1.3.2-1+rpt16), libdrm-amdgpu1:arm64 (2.4.114-1+b1, 2.4.123-1~bpo12+1+rpt1), cups-core-drivers:arm64 (2.4.2-3+deb12u7, 2.4.2-3+deb12u8) [... cut ...] ion4:arm64 (43.2-1, automatic), libyelp0:arm64 (42.2-1, automatic), gnome-terminal-data:arm64 (3.46.8-1, automatic), yelp:arm64 (42.2-1, automatic), gnome-terminal:arm64 (3.46.8-1), fonts-dejavu-extra:arm64 (2.37-6, automatic)
/var/log/apt/history.log.5.gz-End-Date: 2024-10-28  17:17:35
/var/log/apt/history.log.5.gz-
/var/log/apt/history.log.5.gz-Start-Date: 2024-10-29  15:57:20
/var/log/apt/history.log.5.gz-Commandline: apt full-upgrade
/var/log/apt/history.log.5.gz-Requested-By: XXXXXXX (1000)

daverichathome, 2025-03-01:
Also, CLI install of BVM fails on the firstinstal stage.

Botspot, 2025-03-01:
> Also, CLI install of BVM fails on the firstinstal stage.

Please be more specific than that. Can you send some output or a screenshot?

Botspot, 2025-03-01:
Based on your apt install history, it looks like you installed gnome-terminal manually. Do you remember doing that?
Does gnome-terminal launch correctly when running this command?
```
gnome-terminal
```
If my theory is right, that ought to work. Now try this:
```
GDK_BACKEND=x11 gnome-terminal
```

daverichathome, 2025-03-01:
Sorry.  Don't have any screenshot or output.  When it switches to QEMU, it
starts the BVM GUI and that (IMHO) is having the same problem or is
related.  It eventually stops QEMU and returns to the terminal.

daverichathome, 2025-03-01:
gnome-terminal does not start.  Get this error message.  Which I believe is
the same.

Error constructing proxy for
org.gnome.Terminal:/org/gnome/Terminal/Factory0: Error calling
StartServiceByName for org.gnome.Terminal: Timeout was reached

Botspot, 2025-03-01:
> gnome-terminal does not start.  Get this error message.  Which I believe is the same.

Did BOTH of my gnome-terminal commands not work, or did just the second one that had GDK_BACKEND=x11 not work?
Try running these commands in a fresh terminal instance.

daverichathome, 2025-03-01:
In Fresh Terminal ran both commands.  Same error msg.

Botspot, 2025-03-01:
Is your OS fully apt upgraded?

daverichathome, 2025-03-01:
Yep.

0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded

Botspot, 2025-03-01:
What is your current system locale? That could be related. https://github.com/Botspot/pi-apps/issues/2707#issuecomment-2614074998

daverichathome, 2025-03-01:
> What is your current system locale? That could be related. Botspot/pi-apps#2707

LANG=en_US
LANGUAGE=en_US
LC_CTYPE="en_US"
LC_NUMERIC="en_US"
LC_TIME="en_US"
LC_COLLATE="en_US"
LC_MONETARY="en_US"
LC_MESSAGES="en_US"
LC_PAPER="en_US"
LC_NAME="en_US"
LC_ADDRESS="en_US"
LC_TELEPHONE="en_US"
LC_MEASUREMENT="en_US"
LC_IDENTIFICATION="en_US"
LC_ALL=en_US

Botspot, 2025-03-01:
Okay, that's the problem. Seems like gnome-terminal needs a UTF8 locale to launch. I've opened an issue https://github.com/raspberrypi/bookworm-feedback/issues/380

For you, try changing the locale to en_US.UTF-8

daverichathome, 2025-03-01:
> launch. I've opened an issue raspberrypi/bookworm-feedback#380

Will do.  WIll update once that is done and I have validated that it is
working properly.

daverichathome, 2025-03-01:
Pi-Apps installed fine.  BVM seemed to be doing fine.  Then during Step 5
"Install Windows 11", QEMU started, then failed and got this message.  The
XXXXXX is me masking my username.  The correct path is actually displayed.

"Turns out Raspberry and Windows do mix... forming a very crunchy smoothie."
BVM mode firstboot
Windows should install 100% automatically. This will take several hours.
/home/XXXXXX/.local/share/bvm/bvm: line 737: 15185 Killed
 qemu-system-aarch64 "${full_qemu_flags[@]}"
QEMU did not exit successfully.
Operation failed! Please review the errors above.

Botspot, 2025-03-01:
As mentioned on the README.md, the `/home/XXXXXX/.local/share/bvm/bvm: line 737: 15185 Killed qemu-system-aarch64 "${full_qemu_flags[@]}"` message means QEMU was using enough RAM that Linux decided to end the process.

Try it again with no other programs running. Also how much RAM does your device have?

daverichathome, 2025-03-01:
4GB Pi 4.  Will look at config and free up memory.

Botspot, 2025-03-01:
Installing ZRAM will probably fix it. https://pi-apps.io/install-app/install-more-ram-on-raspberry-pi

daverichathome, 2025-03-01:
Thanks.  Already read your notes in the config and installed "more RAM".

Currently in Windows Setup.  Noticeably slower install then my Pi5 w/8GB
and NVME.  But I expected that.

Botspot, 2025-03-01:
Awesome. Comment back here or open a new issue if you run into any more problems please.
Because of your report, I've added a detection for qemu being killed, with a ZRAM recommendation message,

closed 2025-03-01 by Botspot