Botspot/bvm #9: Copy Paste Text Data between Raspberry Pi OS and Windows 11 (QEMU VM) not working

farjanatech, 2025-02-28:
Did you check this issue? **Copy-Paste** Any Text Data between Raspberry Pi OS and Windows 11 (QEMU VM) not working.
Thanks in Advance.

Botspot, 2025-02-28:
Are you using the `boot` mode or the `connect` mode?

farjanatech, 2025-02-28:
Honestly, I have not used your script yet, but I am running Windows 11 ARM using the QEMU emulator in my Raspberry Pi 5 (8GB) model. But copy-paste is not working, as I mentioned here. I see you are also utilizing QEMU. That's why I asked this. Would you please test copy-paste in your test? Here are the commands I use to start my Windows-installed QEMU emulator.

qemu-system-aarch64 \
    -M virt -cpu host -accel kvm -m 4G -smp 4 \
    -device virtio-gpu-pci \
    -bios /usr/share/qemu-efi-aarch64/QEMU_EFI.fd \
    -device qemu-xhci -device usb-kbd -device usb-tablet \
    -drive file=win11.qcow2,if=virtio \
    -nic user,model=virtio-net-pci,mac=00:11:22:33:44:55 \
    -drive file=virtio-win.iso,media=cdrom,if=none,id=inst \
    -device usb-storage,drive=inst \
    -usb -device usb-audio

Botspot, 2025-02-28:
Copy and paste won't work unless you use an RDP connection.
BVM's `connect` mode takes care of this.

farjanatech, 2025-02-28:
That's good! I will remove my existing QEMU setup for Windows 11 ARM and install your script. 
However, I am slightly confused about the setup process. Is there any video tutorial on YouTube, or do you have a plan for a video tutorial?

Botspot, 2025-02-28:
> Is there any video tutorial on YouTube, or do you have a plan for a video tutorial?

Nope, not yet. Some youtubers may make videos about it soon. BVM has only been out for 2 days so far. https://github.com/Botspot/bvm/commit/038a5b452f3b9691cc64355c8272967c17ac2e9a

I suggest you follow the instructions as best you can, and ask any questions right here in this issue if you get stuck! It should be pretty easy.

geerlingguy, 2025-02-28:
> Some youtubers may make videos about it soon.

Working on setting it up right now ;)

closed 2025-03-01 by Botspot