Botspot/bvm #17: Discussion point - preallocating the qcow2 file or not

tinker2much, 2025-03-01:
You're currently preallocating the disk.qcow2 file to its full 40GB size using the "prealloc=falloc" option. I'm wondering whether it would be better to NOT do that, for reasons I'll enumerate, but I would be glad to hear your reasons for doing so. 

if you didn't do so:
1. users could try bvm on smaller SD cards / partitions / disks. 
2. it would be easier to keep several installations around [ if 40GB is big, multiple 40GBs is bigger ;-) ]
3. making backups via simple file copies would be faster and waste less space. 

I don't know how many of the prospective users would be familiar enough with qemu-img to know they could make smaller copies or backups via "qemu-img convert -f qcow2 -O qcow2 disk.qcow2 disk.qcow2.bak001", which is what I'm currently doing.

Botspot, 2025-03-01:
I read in a [benchmarking PDF](https://events19.lfasiallc.com/wp-content/uploads/2017/11/Storage-Performance-Tuning-for-FAST-Virtual-Machines_Fam-Zheng.pdf) that falloc was a lot faster than not preallocating, while it does not actually use disk space.
Right-click on a disk.qcow2 before firstboot and see that the size on disk is a few KB.

Botspot, 2025-03-01:
Another point is that it seems not doing any preallocation still results in about the same filesize on disk, even though windows is not using it. I think I remember it using 32 of the 40GB in terms of filesize, while Windows should ebe able to install on 15GB. (according to my notes on wor-flasher)

So unless I'm missing something, avoiding preallocation would not help, and it would just slow the Windows installation down. A better idea: just set a smaller disk size from the beginning, which can easily be done in the config file.

tinker2much, 2025-03-01:
That article says "Consider enabling preallocation when disk space is not a concern", implying to me that it's actually taking the space, whether allocated via falloc or full???

Botspot, 2025-03-01:
Unless I'm wrong, it seems Windows writes little bits of data to most of the sectors, making the file take up most of that space no matter if it was falloc preallocated or not preallocated.
Pretty sure the size on disk would be near 40 GB in both cases.

What I'm trying to say is that what you are actually looking for is not "no preallocation", but starting with a smaller disk to begin with.

tinker2much, 2025-03-01:
> I read in a benchmarking PDF that falloc was a lot faster than not preallocating, while it does not actually use disk space.
> Right-click on a disk.qcow2 before firstboot and see that the size on disk is a few KB.
> 

Collecting data points on disk sizes by creating them with different options...

I ran just the code in "bvm prepare" that creates the disk.qcow2 file, once as is and once taking out the preallocation=falloc. (I created the two files on a different partition on the same SSD drive, since there's not enough space in the partition I've been working in, and I didn't want to delete the disk.qcow2 file I Windows-updated last night):
```
    #errors="$(qemu-img create -f qcow2 -o cluster_size=2M,nocow=on,preallocation=falloc "$vmdir/disk.qcow2" "${disksize}G" 2>&1)" || error "\nFailed to create $vmdir/disk.qcow2\nErrors:\n$errors"
    errors="$(qemu-img create -f qcow2 -o cluster_size=2M,nocow=on,preallocation=falloc "/media/pi/rootfs5/home/pi/disk.qcow2" "${disksize}G" 2>&1)" || error "\nFailed to create /media/pi/rootfs5/home/pi/disk.qcow2\nErrors:\n$errors"
    errors="$(qemu-img create -f qcow2 -o cluster_size=2M,nocow=on "/media/pi/rootfs5/home/pi/disknp.qcow2" "${disksize}G" 2>&1)" || error "\nFailed to create /media/pi/rootfs5/home/pi/disknp.qcow2\nErrors:\n$errors"

```
An "ls" gives their sizes:
```
pi@piapps:/media/pi/rootfs5/home/pi $ ls -al disk*
-rw-r--r-- 1 pi pi     6291464 Mar  1 07:22 disknp.qcow2
-rw-r--r-- 1 pi pi 42960158720 Mar  1 07:22 disk.qcow2

```
I right-clicked on the original-type file, and I believe it's taking up the full 40GB:

[image]

tinker2much, 2025-03-01:
I tried copying my current Windows-updated disk.qcow2 to a new disk which wasn't pre-allocated, using the following command:
`qemu-img convert -f qcow2 -O qcow2 disk.qcow2 /media/pi/rootfs5/home/pi/diskconv.qcow2`
The result takes up 25GB:
```
pi@piapps:/media/pi/rootfs5/home/pi $ ls -al diskconv.qcow2 
-rw-r--r-- 1 pi pi 25364660224 Mar  1 09:24 diskconv.qcow2

```
I haven't yet made the experiment of doing the full install to a disk which wasn't pre-allocated, given how long that would take.  
I'll probably try swapping the original and copied drives, to verify how the vm runs.

Botspot, 2025-03-01:
It seems you are correct. I do recall having Windows install and watching the size on disk slowly go up after I started using preallocation=falloc, but now can't get it to do that anymore. Not sure what happened, but it could have been due to staying up all night getting this working. :)

Someone needs to benchmark using BVM with preallocation=metadata, and if the final file size is much smaller than 40GB then I will switch to this.

tinker2much, 2025-03-01:
I now trying the firstboot to a disk which wasn't pre-allocated, to see how that works, how long it takes, and what size disk I end up with.

tinker2much, 2025-03-02:
The install to a non-preallocated disk seemed to finish OK, at least I could click and open things from within the Windows desktop, so I shut it down. This message appeared shortly after (I don't see how hibernation is involved - I hope you do).
```
BVM mode firstboot
Windows should install 100% automatically. This will take several hours.
QEMU closed.
Windows is hibernated, refused to mount.
The disk contains an unclean file system (0, 0).
Metadata kept in Windows cache, refused to mount.
Falling back to read-only mount because the NTFS partition is in an
unsafe state. Please resume and shutdown Windows fully (no hibernation
or fast restarting.)
Could not mount read-write, trying read-only
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
It appears that Windows installation was not completed in disk.qcow2. C:/Program Files/Qemu-ga/qemu-ga.exe does not exist.
```
Other facts: 

The firstboot process took longer this way - 2 1/2 hours versus 2 hours. 

Also, the disk.qcow2 file ended up taking 16GB:
`-rw-r--r--  1 pi pi 16049504256 Mar  1 17:53 disk.qcow2`

tinker2much, 2025-03-02:
I'm trying to boot it up again to see what state things are in, and its taking a long time. So, I couldn't unreservedly endorse non-preallocation yet.

Botspot, 2025-03-02:
> I'm trying to boot it up again to see what state things are in, and its taking a long time to connect (> 10 minutes saying "Waiting for RDP service on VM to start listening on port 3389..."). So, I couldn't unreservedly endorse non-preallocation yet.

For times like this try the `boot` mode directly to take RDP out of the equation.

tinker2much, 2025-03-02:
It seems to boot OK, but then gets hung up on shutdown - it gets to a black qemu screen with that one line something like "display output not active", but "bvm boot" doesn't come back, and I eventually Ctrl-C it.

Botspot, 2025-03-02:
I never encountered that issue. Interesting.
Perhaps the disk is just less reliable when not preallocated, as when it expands the linux filesystem would be more prone to splitting it up. I wonder if there would be a way to check that.

Botspot, 2025-03-02:
I've decided to switch to `metadata` preallocation.

[image]

[image]

commit 6c18c484 (2025-03-02): Use metadata preallocation to close #17

Big thanks to tinker2much for proving me wrong here.
closed 2025-03-02 by Botspot