Botspot/bvm #19: mount_qcow2: Partition 4 not found

bp0, 2025-03-01:
Raspberry Pi 5 8GB, running Raspberry Pi OS with all updates from a NVMe SSD via USB.
After a very long time installing windows, there is an error seen below.  
I've tried twice and failed at the same spot twice.
[image]

Botspot, 2025-03-01:
Assuming you installed from Pi-Apps, run this and copy the output here:
```
~/.local/share/bvm/bvm mount ~/win11
```

bp0, 2025-03-01:
`./bvm mount ~/win11`
```
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██                                                      
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███                                                      
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█                                                      
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█                                                      
 █████🭠   🭖██🭡   █▉     🮋█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"Now your Pi can experience the joy of Windows updates!"
BVM mode mount
/dev/nbd0 disconnected
mount_qcow2: Partition 4 not found
```

Also:
`sudo qemu-nbd --connect="/dev/nbd10" ~/win11/disk.qcow2` and then `lsblk`:
```
NAME      MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda         8:0    0 238.5G  0 disk 
├─sda1      8:1    0   512M  0 part /boot/firmware
└─sda2      8:2    0   238G  0 part /
nbd0       43:0    0     0B  0 disk 
nbd1       43:32   0     0B  0 disk 
nbd2       43:64   0     0B  0 disk 
nbd3       43:96   0     0B  0 disk 
nbd4       43:128  0     0B  0 disk 
nbd5       43:160  0     0B  0 disk 
nbd6       43:192  0     0B  0 disk 
nbd7       43:224  0     0B  0 disk 
nbd8       43:256  0     0B  0 disk 
nbd9       43:288  0     0B  0 disk 
nbd10      43:320  0    40G  0 disk 
├─nbd10p1  43:321  0   256M  0 part 
├─nbd10p2  43:322  0   128M  0 part 
├─nbd10p3  43:323  0   128M  0 part 
├─nbd10p4  43:324  0  38.9G  0 part 
└─nbd10p5  43:325  0   618M  0 part 
nbd11      43:352  0     0B  0 disk 
nbd12      43:384  0     0B  0 disk 
nbd13      43:416  0     0B  0 disk 
nbd14      43:448  0     0B  0 disk 
nbd15      43:480  0     0B  0 disk 
```

RonTipper, 2025-03-01:
I have the same problem and I am using NVME

Botspot, 2025-03-01:
@bp0, this is quite strange. We can clearly see that at least on @bp0's machine that partition 4 does exist.
I cnanot reproduce this issue on my Pi. Could I have the opportunity to remote desktop to your system, either @RonTipper or @bp0 to try to resolve this? We would be using [RustDesk installed from Pi-Apps](https://pi-apps.io/install-app/install-rustdesk-on-raspberry-pi/) and communicate via email to avoid having remote connection details shared publicly.

Trubs333, 2025-03-01:
I came here, same error, also nvme

Botspot, 2025-03-01:
Okay, I'm hoping that recent fixes that I just uploaded should solve this.
Could one of you run `bvm boot ~/win11`? Since firstboot failed, it will retry that last part, which should now work as I've given it a few seconds for the partition 4 to appear as it should.

Trubs333, 2025-03-01:
Thanks, that has done the trick for me. Will continue testing 👍

bp0, 2025-03-01:
After git pull, `bvm boot ~/win11` worked. Thanks.

bp0, 2025-03-01:
Tested from scratch, problem seems fixed.

closed 2025-03-01 by bp0

Botspot, 2025-03-01:
Awesome. Enjoy BVM y'all.