Botspot/bvm #24: Install on RPI 5 - 8g 512G nvme

APShark, 2025-03-02:
Install went fairly well, my system did not proceed to the last step. I had to edit a text file after trying three times, in the WIN 11 DIR to advance to the last step. I did receive a Desktop was successfully installed on all occasions.
I had an issue with updating the windows system using WINDOWS UPDATE. Update for Windows 11 Version 24H2 for arm 64-based Systems (KB5051987) (2) Fault code- **_Failed to install on ‎3/‎1/‎2025 - 0x800f0845._** Three other updates were installed successfully. 
**2025-01 Cumulative Update for .NET Framework 3.5 and 4.8.1 for Windows 11, version 24H2 for arm64 (KB5049622**) 

**2025-01 Update for Windows 11 Version 24H2 for ARM64-based Systems (KB5050575) **

**2024-11 Update for Windows 11 Version 24H2 for ARM64-based Systems (KB5048779) **

Besides the last Windows update failing, everything is working well.

Botspot, 2025-03-02:
> Install went fairly well, my system did not proceed to the last step. I had to edit a text file after trying three times, in the WIN 11 DIR to advance to the last step. I did receive a Desktop was successfully installed on all occasions.

What file? And what were the edits?

APShark, 2025-03-03:
gui-steps-complete.txt  i placed a "5" into that file to be able to proceed to the  RUN WINDOWS 11 menu item.

Botspot, 2025-03-03:
Did the OS get stuck and not allow you to initiate a shutdown? If it had done so, then step 5 would have completed I think.

APShark, 2025-03-04:
Yes, everything completed, and it Windows 11 was going to shut down and reboot, but it brought me back to STEP 5, then I edited the text file to trick it into thinking I did complete the next step, then Windows 11 booted normally and updated all but the final update. Before I edited the gui-steps-complete.txt file I tried the installation three times always one step away from booting. Windows 11 runs fine and everything appears to be working. The internet explorer, the sound, OutLook, and Youtube.  I don't understand why it did not write the final step into the gui-steps-complete file. I am very happy at how this is running! I don't understand why the last update failed and all the other updates succeeded. I listed the fault code above, but here it is again " 0x800f0845".

Botspot, 2025-03-04:
BVM should have given you its reason for thinking step 5 did not finish. Did you see anything like that?

APShark, 2025-03-05:
I believe it did read "failed to mount disk.qcow2"

Botspot, 2025-03-05:
> I believe it did read "failed to mount disk.qcow2"

Okay, I'm fairly sure it's fixed now. That whole process turned out to be unreliable, so I made every step retry up to 5 times, which seems to fix it.

Also, did you get your windows updates fixed? I doubt there is anything BVM could do to fix that.

closed 2025-03-07 by Botspot