Botspot/bvm #25: file system type check

zieglerjm, 2025-03-02:
Maybe a ext4 check should be added.  I am assuming the below error is because it is on a NTFS partition?

cp: setting permissions for ‘/mnt/Data/win11/unattended/viogpudo/./viogpudo.sys’: Operation not permitted

Botspot, 2025-03-02:
BVM already does a partition check to not allow FAT* partitions. NTFS should work.
My bigger concern is with how there is a ./ inside the filepath. That does not seem right, and if that ends up needing fixing, it could solve this issue.

Please enable max debug in the config file, run this step again, and send the full output here.

zieglerjm, 2025-03-02:
After switching to an ext4 partition, the download finished without error.  I then copied the folder back to NTFS, but HD didn't mount on 1st boot.  After switching back to ext4, 1st boot worked & install is running.

Botspot, 2025-03-02:
I tested copying my ~/win11 folder to an .iso with a NTFS filesystem on it. It booted fine.
Is it possible your NTFS drive simply had root folder permissions or something like that? BVM runs as the normal user so if it can't write to the folder then that would explain it.