Botspot/bvm #26: Installation not running at all

lukaszsobala, 2025-03-03:
Hello,

I am using Joshua Riek's Ubuntu Noble on Orangepi 3B. Almost all the steps seemed to succeed, but when I run the installation step, 1 core is pinned and nothing ever happens, no filesystem read or write activity. I tried booting the installation ISO on a self-made `virt-manager` KVM machine and it boots fine. But I want the autopilot experience. 

The qemu version is current for Noble so it should be quite recent. 

Any ideas?

Botspot, 2025-03-03:
Please run `cat /proc/cpuinfo` and send the output.

lukaszsobala, 2025-03-03:
@Botspot I think I know where you're coming from, you want to check if there are big and little cores, because KVM doesn't work when mixing them (that's another thing to consider for future versions BTW, I can try it on Rock 5b which has 2 core types and you need to pin to 4-7 for KVM to even run - but emulation does).

Anyway, here's the output, all the same:

```
processor       : 0
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x2
CPU part        : 0xd05
CPU revision    : 0

processor       : 1
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x2
CPU part        : 0xd05
CPU revision    : 0

processor       : 2
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x2
CPU part        : 0xd05
CPU revision    : 0

processor       : 3
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 atomics fphp asimdhp cpuid asimdrdm lrcpc dcpop asimddp
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x2
CPU part        : 0xd05
CPU revision    : 0

Serial          : bcf8f1936cedec19
```

Botspot, 2025-03-03:
> [@Botspot](https://github.com/Botspot) I think I know where you're coming from, you want to check if there are big and little cores, because KVM doesn't work when mixing them 

Not entirely. I knew Orangepi 3B was not big.LITTLE, it has 4 A55 cores, which means BVM needs to download the older Windows 11 release.

However, I had a hunch that the A72-detection was a false positive in this case, because the script checks for the `fphp` CPU capability, which is present on the Raspberry Pi but not previous models.

Thanks to your output, it proves my suspicion correct - here we have an A55 core reporting that is is capable of `fphp`. So I need to find a new way to detect if CPU cores are capable of running the latest Windows 11.
If you have any ideas, do please let me know.

Botspot, 2025-03-03:
OK I think I found a way. AI saves the day once more.

[image]

Will let you know when this is fixed and what to do to get it working. Thanks for the help thus far.

lukaszsobala, 2025-03-03:
No problem. So are A55 cores incapable of `fphp` and i have a misconfigured kernel? I thought all the extensions would be the same on the same arm arch version.

Botspot, 2025-03-03:
> No problem. So are A55 cores incapable of `fphp` and i have a misconfigured kernel? I thought all the extensions would be the same on the same arm arch version.

No, I think it is capable of fphp. I have no idea what fphp means, but in the Pi world at least, the existence of that extension was good enough to choose a Windows version. I'm finding out that fphp has no connection to ARM core version, so it turned out to be a bad determining factor.

Could you try out my function real quick?
```
find_a76_cores() { #Support big.LITTLE cpus such as RK3588. Return 1 if not all cores are equal, list the A76 ones no matter what
  local cores="$(cat /proc/cpuinfo | grep '^CPU part\|^processor\|^$' | tr '\n' '\r' | sed 's/\r\r/\n/g ; s/\r/ /g')"
  
  local a76_cores=''
  local non_a76_cores=''
  
  #list of A76+ core part numbers: https://github.com/bp0/armids/blob/master/arm.ids
  local grep_match='0xd0b\|0xd0c\|0xd0d\|0xd13\|0xd20\|0xd21\|0xd4a'
  
  a76_cores="$(echo "$cores" | grep -w "$grep_match" | awk '{print $3}')"
  non_a76_cores="$(echo "$cores" | grep -vw "$grep_match" | awk '{print $3}')"
  
  echo "$a76_cores"
  if [ -z "$a76_cores" ] || [ -z "$non_a76_cores" ];then
    #all the cores are the same
    return 0
  else
    #list the cores that are A76
    return 1
  fi
}
find_a76_cores
echo $?
```
It should not give any output and return 0.

lukaszsobala, 2025-03-03:
How do I try it? I am not a programmer and don't know which language this is.

Botspot, 2025-03-03:
Yes, just paste all of that into a linux terminal and press enter.

Botspot, 2025-03-03:
We are looking for this output: `0`

Botspot, 2025-03-03:
@lukaszsobala are you still there?

lukaszsobala, 2025-03-03:
returns 
`0`

lukaszsobala, 2025-03-03:
on Rock 5B returns:
```
4
5
6
7
1
```

lukaszsobala, 2025-03-03:
@Botspot I installed BVM through pi-apps, any way to update it there? Should I reinstall?

Botspot, 2025-03-03:
> [@Botspot](https://github.com/Botspot) I installed BVM through pi-apps, any way to update it there? Should I reinstall?

No need. BVM auto-updates on every run.

lukaszsobala, 2025-03-03:
this did not fix my issue. Still, 1 core is pinned at 100%, no write activity and nothing happens. The Windows version downloaded seems to be the same (24H2). Please reopen the issue.

[image]

Botspot, 2025-03-03:
Just found the remaining issue.

Botspot, 2025-03-03:
Alright, should be all fixed now with most recent change https://github.com/Botspot/bvm/commit/39ed66dad105a5de227561193ae4fc9ecdc3c2de

As before, just start again with the download mode.

You should look for a warning: "Downloading an older Windows version that is compatible with your device. Later versions of Windows 11 require ARM instructions that your CPU lacks." This means it's working correctly now.

lukaszsobala, 2025-03-03:
I modified my executable to use AAVMF instead and it still doesn't work, but the RAM usage is a bit higher, so something more seems to have happened. Still, this did not solve the issue.

Botspot, 2025-03-03:
Reopening for now. I'm surprised - I thought this would work due to how Orange Pi 5 users say it runs great.
I think this is beyond my ability to make suggestions at this point. Let's arrange a time to set up a remote desktop connection to give me a chance to iterate changes until it works.

lukaszsobala, 2025-03-03:
Let me know if I can assist. I can send a virt-manager xml file which boots, at least from the installation CD. One of the key things to disable that qemu enables by default were hyperv features: relaxed, vapic, spinlocks. I have no clue what these are for but with them enabled by default, the machine won't even start.

I would also try with PAN_MESA_DEBUG=gl3, maybe this is the problem (should I just export the env at the beginning of the BVM script?)

lukaszsobala, 2025-03-03:
heyy @Botspot that did it! with gl3 things are happening!

Botspot, 2025-03-03:
Okay, what exactly did you set? Just `export PAN_MESA_DEBUG=gl3` before running bvm fixes it? I want to add this to the script if there are no bad side effects.

> But please try reverting to 24H2.

Before I revert the change, please try 24h2 to see if it actually gets past the BIOS screen. If it does, then it means the people saying the latest Windows 11 needs an A76 CPU are misinformed and mistaking correlation for causation. 

To try 24h2, just replace the `find_a76_cores` function to force-report the CPU is A76.
Like this:
```
find_a76_cores() {
echo -e '0\n1\n2\n3'
return 0
}
```
And then run bvm in the download mode again. To prevent your temporary function change from being overwritten, change the config file to disable updates.

lukaszsobala, 2025-03-03:
I added `export PAN_MESA_DEBUG=gl3` just below the bash shebang.

lukaszsobala, 2025-03-04:
The installation step went OK but then the VM refused to boot. After restarting, it refused to boot again. I then tried booting from virt-manager and it went well. I am now updating Windows, possibly to 24H2.

But now I can't modify bvm because every time it launches it checks Internet connectivity, and if it's off, it does not proceed booting Windows.

Botspot, 2025-03-04:
After talking to a developer who knows more about Windows' latest CPU requirements, I have updated BVM to check for the 'atomics' feature when determining which ISO to download, and have clarified the difference in how performance cores are detected in light of that.
Also added `export PAN_MESA_DEBUG=gl3` to BVM in the same commit.

> The installation step went OK but then the VM refused to boot. After restarting, it refused to boot again. I then tried booting from virt-manager and it went well. I am now updating Windows, possibly to 24H2.

Again, I'm still interested in helping troubleshoot by seeing your screen and iterating over qemu commands until it works reliably. It could be related to how `firstboot` uses ramfb, but `boot` uses virtio-gpu-pci.

lukaszsobala, 2025-03-04:
Oh, you meant you wanted to set up remote access to me? I did not invert this from the previous comment, I thought you have some kind of access already.

Botspot, 2025-03-04:
Looks like we are both misunderstanding each other. No I own no Orange Pi devices, so that's why I'm asking to test things on yours. Is RustDesk installing OK? Once it installs please run it and send the id number. Don't send the password.

lukaszsobala, 2025-03-04:
I can't do it now. Will reply when I can.

lukaszsobala, 2025-03-04:
`PAN_MESA_DEBUG=gl3` fixed the installation and running steps. Windows 24H2 works OK on core A55 too (they have `atomic`).

commit 40a0e5bc (2025-03-03): Improve find_a76_cores accuracy, use 2 unattended.iso, add tooltips

Closes #26
closes #28
commit 39ed66da (2025-03-03): use find_a76_cores to download correct ISO to close #26
closed 2025-03-04 by lukaszsobala