Botspot/bvm #30: Operation failed due to using too much RAM

jboots07, 2025-03-03:
As title states, the install was taking forever on step 5, Windows 11 install (step 5 in GUI) was on the 2nd step. Let run overnight and saw a message in the window saying QEMU was using too much ram and to enable ZRAM, however ZRAM is already enabled. I am running the install again to see if it fails a second time, but any other suggestions?

Pi 4 2GB model, 256GB Samsung SD Card

Botspot, 2025-03-03:
I don't have a 2GB Pi model, but in making BVM I knew that anyone using 2GB is really going to be pushing limits.

- Try killing `pcmanfm` - it closes your file manager and should save 100-300 MB of RAM.
- Try switching to x11. This may use less RAM.
- Try lowering the RAM amount allocated to the VM:
    ```
    #Force RAM allocation for QEMU. (in gigabytes) It's recommended to leave this commented for BVM to choose a sensible value.
    vm_mem=1.9
    ```
    BVM uses 3GB by default on the 2GB Pi, which is pushing the limits. Maybe start with 1.9 and slowly go down from there.

Please let me know how it goes. With these tips combined, it might be possible to get it working, though no promises.

jboots07, 2025-03-03:
> I don't have a 2GB Pi model, but in making BVM I knew that anyone using 2GB is really going to be pushing limits.
> 
> * Try killing `pcmanfm` - it closes your file manager and should save 100-300 MB of RAM.
> * Try switching to x11. This may use less RAM.
> * Try lowering the RAM amount allocated to the VM:
>   ```
>   #Force RAM allocation for QEMU. (in gigabytes) It's recommended to leave this commented for BVM to choose a sensible value.
>   vm_mem=1.9
>   ```
>   
>   
>       
>         
>       
>   
>         
>       
>   
>       
>     
>   BVM uses 2GB by default on the 2GB Pi, which is pushing the limits, but too far lower than that and the installation fails. Maybe start with 1.9 and slowly go down from there.
> 
> Please let me know how it goes. With these tips combined, it might be possible to get it working, though no promises.

TY for the reply! I am trying the install again, then will try the tips you suggested if it fails. I have other Pi's, but just had this one as a spare not being used and figured, hey why not give it a try :)

htop screen shot: https://imgur.com/VA8L6SH

jboots07, 2025-03-03:
I mis clicked, didn't mean to close this. Changed to X11. I disabled pcmanfm, however I lost desktop GUI, which I would rather use. How do I change the RAM allocation? It looks like per htop my Pi has 1.80GB of RAM, is it advisable to drop below that then?

Botspot, 2025-03-04:
> I mis clicked, didn't mean to close this. Changed to X11. I disabled pcmanfm, however I lost desktop GUI, which I would rather use. 

I think Windows can use less RAM once it's installed, so this should be temporary. pcmanfm is non-essential, so I suggested it as a way to free up a little RAM.

> How do I change the RAM allocation? It looks like per htop my Pi has 1.80GB of RAM, is it advisable to drop below that then?

Yes, it's an option in the config file.

jboots07, 2025-03-04:
It still failed even with 1.6GB set. I think it's safe to say this won't work on 2GB models at this point. I have a pi 5 on the way and will install on that once it arrives.

Botspot, 2025-03-04:
> It still failed even with 1.6GB set. I think it's safe to say this won't work on 2GB models at this point. I have a pi 5 on the way and will install on that once it arrives.

It's not likely to be very practical, but if you are down to test, it may work. I want to be able to confidently say the 2GB model is not enough, if true.
At some point, QEMU should stop being killed and Windows should start saying it failed to install. I know I got to that point when setting 1 GB in tests. Have you reached that point yet, or is QEMU still being killed due to its RAM usage?

jboots07, 2025-03-04:
> It's not likely to be very practical, but if you are down to test, it may work. I want to be able to confidently say the 2GB model is not enough, if true. At some point, QEMU should stop being [...]

QEMU is always killed.

jboots07, 2025-03-04:
But yes, I am down to test further if you have any other suggestions!

jboots07, 2025-03-05:
Do you think optimizing the usage of ZRAM could help?

[zram0]
zram-size = ram * 3 / 4
compression-algorithm = zstd
swap-priority = 110

echo "vm.swappiness=80" | sudo tee -a /etc/sysctl.conf

Botspot, 2025-03-05:
> Do you think optimizing the usage of ZRAM could help?
> 
> [zram0] zram-size = ram * 3 / 4 compression-algorithm = zstd swap-priority = 110
> 
> echo "vm.swappiness=80" | sudo tee -a /etc/sysctl.conf

Pi-Apps already does those. See https://github.com/Botspot/pi-apps/blob/master/apps/More%20RAM/install

Botspot, 2025-03-06:
You could try the Tiny11 ARM installer. Maybe that requires less RAM. https://archive.org/details/tiny11a64
Be aware I have not tried it, but if you choose to download it, rename it to installer.iso and replace the existing one in ~/win11, then do `firstboot` again.

jboots07, 2025-03-06:
> You could try the Tiny11 ARM installer. Maybe that requires less RAM. https://archive.org/details/tiny11a64 Be aware I have not tried it, but if you choose to download it, rename it to installer.iso and replace the existing one in ~/win11, then do `firstboot` again.

Are you able to share with me some more details about 'firstboot'? I started over on step 4 after replacing the ISO, which may not be the correct step, then clicked and started step 5 in the GUI. Now the install seems to not be unattended and asks for an install drive but subsequently is unable to find one.

Botspot, 2025-03-06:
Another person on my Discord tried Tiny11 and seemed to have the same result. Something about tiny1 breaks driver installation.
You are welcome to try manually installing drivers by browsing for drivers on the E: drive, but this other user said the driver installation failed.