Botspot/bvm #32: Modify C: drive partition size

Trubs333, 2025-03-03:
Where or when do I modify the config to change the 40G default?

I did it post install but it was cumbersome mounting and resizing the partition.

Botspot, 2025-03-03:
The config value is applied during the `prepare` mode. Changing it later won't have any effect, at least not yet.

Trubs333, 2025-03-03:
Okay many thanks

So this

 <--- Please read the config file!

Could be interpreted as this

 <--- Please read the config file _and make any required adjustments_

Would be my understanding

closed 2025-03-04 by Trubs333

Botspot, 2025-03-04:
Soon after you closed this issue, I finished adding an all-new `expand` mode. commit https://github.com/Botspot/bvm/commit/856dd0eaff6e4ea10038647cb189d3f9c5604b92

I also added a convenient button for it.

Trubs333, 2025-03-04:
That is really useful thanks

Botspot, 2025-03-04:
> That is really useful thanks

To be honest, the only reason I got it done so soon was because I ran out of space in the VM so had to figure it out. 😂