Botspot/bvm #37: USB Passthrough shows USB hub but won't recognize any USB devices

vjkanth, 2025-03-03:
The config file was altered to include all USB devices as listed by lsusb command, ensuring each set is separated by a space character and commenting hash removed. When windows installed and booted ok, under device manager I can see the USB hub listed. However no actual devices such as pen drives or dongles are being recognized. I've tried multiple reboots and full reinstallations to no avail.

My setup:
RPi 5 4GB, NVME 256GB
Pi OS 64 Bit (bookworm)
PiApps installed
BVM installed through PiApps
Has a USB Sound DAC hardware attached to A USB 2.0 port and works well in Win VM.
USB Mouse and Keyboard connected to USB 3.0 ports work well. I connected these through a USB hub and are usable.
I've tried using a variety of pen drives, and whilst the host OS recognizes them, the Windows VM doesn't. Any ideas?

Botspot, 2025-03-03:
I should mention this in the config file: usb hubs cannot be forwarder, only usb endpoint devices such as pen drives and webcams work.

One other user had an issue with usb passthrough and udev ended up being somehow messed up in an unclear way.
Try running this, and press and hold the Page Down key until you reach the bottom, then see if there are any yellow warnings.

```
sudo journalctl -u systemd-udevd
```
Please copy what you see here, making the terminal a larger size if necessary.

vjkanth, 2025-03-04:
First and foremost, thank you for your prompt response and also this fabulous project. 
When attempting to get the log entries today as suggested, I noticed that I got 3 new options on the BVM Main window, mainly the USB Passthrough option:

[image]

Selecting the USB Passthrough option gives me the same list of USB devices that I'd added in the Config file during BVM setup. I just had to select the Pen Drive and as you can see from the screenshot, it now shows up on Win11. I also tried unplugging and replugging between the connected USB Hub and directly on the PI's USB 2.0 port and both ways work flawlessly. 

One observation, the USB Mouse and Keyboard work in the BVM regardless of my selecting the two from the USB Passthrough options:

[image]

I mainly setup the BVM to run SDR radio using Windows application SDR Uno working with SDRPlay RSP1 and I can confirm that using the method above I have managed to run SDR Uno with RSP1 albeit with some stuttering perhaps due to system resource limitations(?). Need to investigate this more but it's looking hugely positive.

I'm very pleased to see this and thank you and all supporting contributors for this brilliant solution. I've been looking for something better than Wine and Box86 for my purpose and this is certainly it! 

Given the above this "issue" may be closed, but would recommend adding some instructions around the USB Passthrough menu option on the main BVM Window with perhaps a screenshot.

closed 2025-03-04 by vjkanth