Botspot/bvm #41: Using Windows 10

Nuggets1877, 2025-03-04:
Hello, i've come here to ask for help installing Windows 10 on BVM because, as we can ser on the description, botspot says that is possible. I just wanted to know how because i love Windows 10 and Im using a pi 4 model b, and Windows 10 Will Run better. Hope that someone could help me.

Botspot, 2025-03-05:
Update: I've added a feature that allows you to skip just the installer.iso download if one already exists at the right location.

Botspot, 2025-03-05:
Delete image.esd, there is no need for it. If the process completed, there would be a installer.iso in the very same folder.
You should be able to put your windows 10 iso in the vm directory you are using, just making sure to name it installer.iso. Then run BVM with the `download` mode and if the file is in the right place it will ask if you want to replace it. Answer no. (press n, press enter)

Nuggets1877, 2025-03-05:
> Delete image.esd, there is no need for it. If the process completed, there would be a installer.iso in the very same folder. You should be able to put your windows 10 iso in the vm directory you are [...]

So, first i Run the download step, then i replace the installer.iso(i've already found out) and then i Run the download step again or continue to the Next step?

Botspot, 2025-03-05:
1. Before my fix, you would have needed to run the download step first, then delete the installer.iso that it made and replace it with your own, and not run the download mode again.
2. Now, you can run the download mode after moving the installer.iso to the right place, and now the download mode will offer the option to skip downloading Windows but still download everything else.

If the download step already completed, no need to run it again.

Nuggets1877, 2025-03-05:
Installed Windows but then it gave this error[

[image]

](url)

Botspot, 2025-03-05:
Huh that's interesting. Well if it got that far, it may have installed most of the way, or maybe all of the way.
Try booting it with the `boot` mode.

Nuggets1877, 2025-03-06:
> Huh that's interesting. Well if it got that far, it may have installed most of the way, or maybe all of the way. Try booting it with the `boot` mode.

And no, that didn't worked, i even tried to repeat the process from the beggining and its the same. And the problem os not with the iso, because its from uup dump, that one witch generates isso from CMD files, but i noticed that the Windows 10 ISO, the word ISO is on capital letters and the installer .iso not and i had to Change that. Btw is Windows 10 Faster than 11?

Nuggets1877, 2025-03-08:
> And no, that didn't worked, i even tried to repeat the process from the beggining and its the same. And the problem os not with the iso, because its from uup dump, that one witch generates isso from [...]

Any news?