Botspot/bvm #47: BVM will not boot

Botspot, 2025-03-05:
### Discussed in https://github.com/Botspot/bvm/discussions/46

<div type='discussions-op-text'>

<sup>Originally posted by **Jdubaya** March  4, 2025</sup>
I hope this is the correct spot to ask this.

I have installed BVM on mi 8GB rpi5 through pi apps.
Installation went fine, and BVM did boot up and run on the first run.

Now when I try to start BVM (direct or headless) the terminal shows booting, and then stops with an error saying qemu does not seem to be running. Maybe the OS failed to boot.

I have not changed anything, and am unsure what the issue could be.</div>

@Jdubaya let's discuss this here in an issue please.

Botspot, 2025-03-05:
@Jdubaya could you try enabling both debug options in the config file, then running the boot mode again?

Jdubaya, 2025-03-06:
I re-installed. It is working now.
Now I just have to figure out usb connections.

Botspot, 2025-03-06:
Usually it's better to find the cause of an issue rather than reinstall to cover up the original problem, but I'm glad it is working for you now. Open a new issue/discussion if you need more help!

closed 2025-03-06 by Botspot

GamerPlayz-Gaming, 2025-03-24:
This is me right now too