Botspot/bvm #50: No Window Decorations using RDP on Ubuntu 24.04

ItsAConspiracy, 2025-03-05:
Window decorations are missing when using `bvm connect ~/win11 `
In gnome there is no way to move, resize or maximize that window at all. Its like the "-decorations" flag is set. 
Setting the flag "+window-drag" doesnt appear to help either. ALT + Left-Click-Drag also does not move the window.
[image]

This is an Orange Pi 5 running Ubuntu 24.04

Botspot, 2025-03-05:
Someone else encountered this, also on ubuntu and on a Orange Pi 5. It does not seem to happen anywhere else. I already had them try running a minimal wlfreerdp command with as few options as possible, and also run with `+decorations`, neither of which made a difference.
Could you share all output in the terminal that's running the `connect` mode?

What seemed to fix it for that other user was to switch to viewing the VM with remmina. Ideally freerdp would work though, as it's a bit faster.

ItsAConspiracy, 2025-03-06:
Yes I had tried `+decorations` also to no effect. Here is terminal output for `bvm connect ~/win11`
```
BVM mode connect
Waiting for RDP service on VM to start listening on port 3390... Done
[19:13:37:926] [3332:3332] [WARN][com.freerdp.core.gcc] - Server uses non-advertised encryption method 0x00000000
[19:13:38:032] [3332:3332] [INFO][com.freerdp.gdi] - Local framebuffer format  PIXEL_FORMAT_BGRA32
[19:13:38:032] [3332:3332] [INFO][com.freerdp.gdi] - Remote framebuffer format PIXEL_FORMAT_BGRA32
[19:13:40:252] [3332:3435] [INFO][com.freerdp.channels.rdpdr.client] - Loading device service drive [gvfs] (static)
[19:13:40:253] [3332:3435] [INFO][com.freerdp.channels.rdpdr.client] - Loading device service drive [BU] (static)
[19:13:40:254] [3332:3435] [INFO][com.freerdp.channels.rdpdr.client] - Loading device service drive [home] (static)
[19:13:40:268] [3332:3332] [INFO][com.freerdp.channels.rdpsnd.client] - [static] Loaded pulse backend for rdpsnd
[19:13:40:268] [3332:3332] [INFO][com.freerdp.channels.drdynvc.client] - Loading Dynamic Virtual Channel rdpsnd
[19:13:40:268] [3332:3332] [INFO][com.freerdp.channels.drdynvc.client] - Loading Dynamic Virtual Channel rdpei
[19:13:40:268] [3332:3332] [INFO][com.freerdp.channels.drdynvc.client] - Loading Dynamic Virtual Channel rdpgfx
[19:13:40:269] [3332:3332] [INFO][com.freerdp.channels.drdynvc.client] - Loading Dynamic Virtual Channel disp
[19:13:40:614] [3332:3445] [INFO][com.freerdp.channels.rdpsnd.client] - [dynamic] Loaded pulse backend for rdpsnd
[19:13:43:023] [3332:3445] [INFO][com.freerdp.channels.rdpsnd.client] - [dynamic] Loaded pulse backend for rdpsnd
[19:13:43:302] [3332:3332] [WARN][com.freerdp.core.rdp] - pduType PDU_TYPE_DATA not properly parsed, 562 bytes remaining unhandled. Skipping.
[19:13:43:399] [3332:3435] [INFO][com.freerdp.channels.rdpdr.client] - registered device #1: gvfs (type=8 id=1)
[19:13:43:403] [3332:3435] [INFO][com.freerdp.channels.rdpdr.client] - registered device #2: BU (type=8 id=2)
[19:13:43:403] [3332:3435] [INFO][com.freerdp.channels.rdpdr.client] - registered device #3: home (type=8 id=3)
```
At least I dont think its gnome specific since this guy is using a different window manager: https://www.youtube.com/watch?v=X5xyc-z3BwU but yeah, its another Orange Pi 5.

Lolmc0587, 2025-03-06:
i have the same issue with freerdp on Orange 5 Plus

Lolmc0587, 2025-03-06:
xfreerdp on wayland seem work fine

closed 2025-03-06 by Botspot

Botspot, 2025-03-06:
I found some ways to improve Remmina's performance and desktop integration, and it's now the new default for `connect`.

Any new issues caused by this need to be reported to me. If everything is fine, then I may eventually remove freerdp from BVM and make remmina the only connect mode.

Lolmc0587, 2025-03-26:
@Botspot  Can you change wlfreerdp to xfreerdp on wayland environment? Remmina seem can't forward touchevent

commit 4650cd52 (2025-03-06): Switch to Remmina for connect, add fullscreen option

Fixes #48 to no longer logout of Windows after being disconnected for 60
seconds (only applies to new installs)
Fixes non passwordless sudo systems trying to install any new
dependencies in a non-visible shell
Fixes usb forwarding udev file not being deleted if usb_passthrough
variable is commented
Changes WM_CLASS for yad to be bvm
Closes #50
After these Remmina optimizations, it runs just as fast as freerdp, but
more reliably and sets window decorations. In a future commit I may
remove freerdp completely.
Restructure config file to be mor [... cut]