Botspot/bvm #53: USB passthrough and win11 drivers

Jdubaya, 2025-03-07:
I am able to forward USB from the Pi5 to BVM.
In this case Silicon Labs Dual cp2105 USB to UART com ports.
These allow cat command and audio to and from my radio transceiver.
I have installed the drivers on the win11 side, but it does not add the audio devices.

Will these USB devices recognize the drivers in win11, or does the passthrough only work with endpoint hardware?

Botspot, 2025-03-07:
Two things could be going wrong here.
1. The RDP connection overrides the OS sound settings, making other audio devices inaccessible. I wrote about this in #35 (search for the word "microphone") with a solution.
2. The device drivers may be x64-only and not be compatible with ARM, in which case you are out of luck. Microsoft says [here](https://learn.microsoft.com/en-us/windows/arm/apps-on-arm-x86-emulation) that emulation is not possible for kernel mode components such as hardware drivers, and such components must be the native architecture.
3. It may be necessary to set up QEMU with serial passthrough instead of usb passthrough in your specific case. This is an area I am unfamiliar with, and lack the hardware to test with.

Jdubaya, 2025-03-07:
Apologies, I identified the wrong usb connections. The Silicon Labs are working as intended.
It seems the usb audio codec is Texas Instruments PCM2903c. 

I tried direct boot, and everything is working just beautifully.
Thank you and great work on this!

This is currently the only way I can filter my audio with AI on rpi.

Botspot, 2025-03-07:
> This is currently the only way I can filter my audio with AI on rpi.

It's good to know you got it working. That's one of the more interesting uses I have seen so far. If you feel so inclined, feel free to describe your setup and workflow a little bit in the show and tell. https://github.com/Botspot/bvm/discussions/categories/show-and-tell

closed 2025-03-07 by Botspot