Botspot/bvm #55: Pi 5 (D0-Stepping) with 16gb RAM / Tastatur

mtcgit09, 2025-03-07:
Hi, I installed Bookworm on a Pi5 (D0 stepping) with 16GB RAM and then (thanks to the excellent work of Botspot) installed Windows 11 Pro (with bloatware).

Sometimes it happens that (logged in to Win11) the keyboard only allows a few keys. When using the scroll wheel on the mouse, various URLs are displayed, which I can then select.

However, I cannot enter numbers or letters using the keyboard. Keyboard and mouse changed ... the same reaction

So far I have not been able to reproduce a trigger for this behavior.

Another topic:
I would like to login without the autologon.

I deleted the installation-user and activated the internal admin. I tried to turn off the autologon via registry ... not working ... only when i'am using "direct" for login, I get the classic login screen. (User/PW)

It would be very nice if someone could give me a helpful information  (I'm not a programmer).

Regards Mike

Botspot, 2025-03-07:
> Sometimes it happens that (logged in to Win11) the keyboard only allows a few keys. When using the scroll wheel on the mouse, various URLs are displayed, which I can then select.

Interesting. Are you usb-forwarding the keyboard or not? Could you use the [botspot-screen-recorder](https://github.com/Botspot/botspot-screen-recorder) tool to record what you are experiencing? The videos should be able to upload straight to a github comment here.

> I tried to turn off the autologon via registry ... not working ... only when i'am using "direct" for login, I get the classic login screen. (User/PW)

This is likely related to how RDP authenticates. I suggest you experiment with using Remmina directly, and research how to connect unauthenticated, to allow the user to input their own username/password. It may also be related to how autounattend.xml sets up the user account - it enables autologin. Not sure how those settings transfer or if it's not possible to change them later.

> It would be very nice if someone could give me a helpful information (I'm not a programmer).

I can try, but on the knowledge of Windows settings and policies, I'm not as good as ChatGPT. :)

mtcgit09, 2025-03-08:
Hi Botspot, I did the installation quite "normally" (via your GUI). I only changed the default VM-user and PW in the config-file, set the language to "German" and set debloat to "false".
The "strange" behavior occurred for the first time immediately after the installation.
I will pay attention to how and whether I can provoke the behavior and then record the effects via botspot-screen-recorder.

I've tried a lot regarding "autologon" (on the windows-side) ... it looks like the registry entries are somehow being overwritten.

I'll experiment a little more...

but ... thank you very much for your help and your excellent work!

Mike

Botspot, 2025-03-08:
> I've tried a lot regarding "autologon" (on the windows-side) ... it looks like the registry entries are somehow being overwritten.

I think your registry changes are working, otherwise this would not happen:

> only when i'am using "direct" for login, I get the classic login screen. (User/PW)

If autologin was still enabled, then it would login automatically in the "direct" mode. Again, I think what is happening is RDP and autologin. Remmina is a tool used for connecting to Windows, and it provides the username and password to Windows on launch, meaning that even with autologin disabled, Remmina immediately logs in when connecting. I don't know if this can be changed or disabled. RDP may have a mode where you connect unauthenticated, then allow the user to choose a user and login with username and password. Most likely your next step is to try launching Remmina directly, apart from BVM, but while the `boot-nodisplay` mode is running, and experiment with all the Remmina advanced options with its graphical interface.

mtcgit09, 2025-03-09:
Hi Botspot, I have now (almost) found out why the keyboard reacts "strangely" (as described above) ...

I noticed that the change definitely has something to do with switching the tasks (using alt+tab multiple times).

If the behavior has occurred once, I can restore the correct state by using alt+tab "multiple" times again.

Now I still have to find out whether the number of times I use alt+tab plays a role or whether this activates eventually a hotkey, that I don't know about.

Regards, Mike

mtcgit09, 2025-03-09:
I have found the solution ... so simple...

When using Alt+Tab to switch tasks, the use of the Alt key is saved, which is not the case in a "pure" Windows environment.

Or is it always like this when working with the Pi (Raspian)?

I have only been working with a kind of Linux environment for a few weeks ... sorry

That means if I am working in the Firefox browser and use Alt+Tab (this brings the BVM window that is still open to the foreground, for example), then when I use Alt+Tab again (so I am back in Firefox), the Alt key remains active... for whatever reason...

If I then press a key (e.g. "H"), the help-pulldown-menue is called up.

A simple press of the Alt key again and you can carry on working normally.

Maybe it's just me ... Pi5 with D0-stepping??? But I can't imagine that ...

Regards Mike

Botspot, 2025-03-09:
> Maybe it's just me ...

Or maybe it's just me... never learning to use the alt key. I'm going to try it right now.

I am going to guess that your keyboard issues will all go away if you usb passthrough the keyboard. Linux will loses its control over the keyboard and all keyboard events go directly to Windows, including Alt+F4, Ctrl+Alt+Delete, etc.

Botspot, 2025-03-09:
> I'm going to try it right now.

I tried switching between typing in a linux text editor, and typing in Windows Notepad, switching with alt-tab presses, but nothing interesting ever happened. In any case, I highly doubt BVM could do anything to solve this issue for you. I know some QEMU power users get 2 keyboards and usb-passthrough one keyboard to windows, leaving the other one connected to linux. You might want to try that if this issue keeps happening and you use this a lot.

mtcgit09, 2025-03-10:
Hi Botspot, you're right, it's better to activate USB passthrough for keyboard and mouse. That way, "irregularities" are eliminated.

That leads to an either/or usage behavior.
Either I work with Win11 or with Raspian, which I'm quite happy with.

Due to my registry changes (installation user deleted, internal administrator activated, ...) only one option for starting Win 11 is "ok" for me ...

USB passthrough: mouse + keyboard are activated
Viewing mode: direct fullscreen

... I would be interested to know whether this option can be saved and possibly started with just one click (icon on the Raspian desktop).

Regards, Mike

Botspot, 2025-03-10:
> ... I would be interested to know whether this option can be saved and possibly started with just one click (icon on the Raspian desktop).

Sure.
- USB passthrough settings have already been saved to `bvm-config` from the GUI, so they don't need to be specified.
- Fullscreen needs to be specified in the config file, and for CLI usage you need to uncomment that line yourself.
- Booting "direct" translates to `bvm boot-gtk ~/win11`.

Simply make a menu launcher that runs that command and it should work.

closed 2025-03-11 by Botspot