Botspot/bvm #56: Need the feature to set a Fixed Custom Resolution for Windows running inside BVM

farjanatech, 2025-03-08:
Dear Sir,
If I can set a Fixed Custom Resolution for Windows, that would be great instead of using the flexible resolution that it currently offers.
For example, 1440x900  or 1280x800, or 1024x768, etc. 
I think a smaller window resolution can perform faster than a full flexible Resolution. A few days ago, I tried many methods to set it using ChatGPT also but failed to set such a Custom Resolution. Finally, I found a third-party tool (Custom resolution setup software: https://www.monitortests.com/download/cru/cru-1.5.2.zip). It helps to set Fixed Custom Resolution for Windows in QEMU.
But without using the third-party tool, can you add this feature directly in BVM?

Next question: In BVM connect what is the difference between Remmina connect and Direct connect?
1) Does **Direct Connect** perform faster than **Remmina Connect**?
2) Does **Direct Connect** support Clipboard Sync (Copy-Paste)?
3) Does **Direct Connect** support **USB Passthrough** and volume? 

Last Query:
Suppose I was working on an MS Word file inside BVM and suddenly closed the BVM window and clicked shutdown on my main Pi 5.
In this situation, if I restart my Pi 5 and re-open the BVM, then will the previous MS Word file start from where it stopped, or will Windows load it as a corrupted and not saved file?

Thanks in Advance
Farjana

Botspot, 2025-03-08:
> Dear Sir, If I can set a Fixed Custom Resolution for Windows, that would be great instead of using the flexible resolution that it currently offers. For example, 1440x900 or 1280x800, or 1024x768, etc.

This is easy to do with at least remmina. Simply edit ~/win11/connect.remmina and change `scale=2` to `scale=1`, then optionally, set any starting resolution you want by changing `window_width` and `window_height`. All config options are documented here: https://remmina.gitlab.io/remminadoc.gitlab.io/md__builds__remmina_remmina_ci__remmina_wiki__remmina__config__file__options.html

> Next question: In BVM connect what is the difference between Remmina connect and Direct connect?
> 
> 1. Does **Direct Connect** perform faster than **Remmina Connect**?

Try it and see. "Direct" uses QEMU to display a "physical" screen from the VM, while FreeRDP and Remmina are RDP clients that connect to Windows using a localhost network connection. I personally think Remmina is the fastest most of the time, but it depends.

> 2. Does **Direct Connect** support Clipboard Sync (Copy-Paste)?
  
"Direct" does not. This is a simple display passthrough using QEMU, and there is no clipboard synchronization.
  
> 3. Does **Direct Connect** support **USB Passthrough** and volume?
All modes support USB passthrough.

> Last Query: Suppose I was working on an MS Word file inside BVM and suddenly closed the BVM window and clicked shutdown on my main Pi 5. In this situation, if I restart my Pi 5 and re-open the BVM, [...]

Windows would definitely not relaunch with Microsoft Word open where you left off, but I don't think your unsaved work would be unrecoverable either. I think MS has a pretty good document recovery feature.
I have experimented with unsafe shutdowns on Windows, to see if it ever gets corrupted. In other words, when I want to stop the VM, rather than clicking shutdown in windows, I have been closing the terminal or pressing Ctrl+C in the terminal to force QEMU to immediately stop. I have done that at least 50 times, and one time Windows had to repair itself automatically on startup, and that was it. It still works fine.

farjanatech, 2025-03-10:
On the first try, your suggestion did not work for the Remmina custom Resolution setting. Once I set `viewmode=1` with your suggestion, it worked. 
**1 = Normal Window
2 = Scaled Window
3 = Fullscreen
4 = Seamless**

closed 2025-03-10 by farjanatech