Botspot/bvm #58: Running on ClockworkPi uConsole.

GregE777, 2025-03-10:
Just an FYI, a few of us are trying this on ClockworkPi uConsole devices with Rpi cm4. Seems to mostly work but I broke it during a Windows cumulative update and ended up testing and starting again. I'm trying to get a couple windows specific apps to run on this device because the form factor is easy to carry around my work.

Thread is here https://forum.clockworkpi.com/t/running-a-hypervisor/15859/24

Next attempt I think I'm going to try the script from Chris Titus to slim it down more and lock updates out. Not the best for security, but I'm also not running it all day long. Might work on this in a couple hours.

Botspot, 2025-03-10:
I have used BVM on my Pi5 for installing windows 3 separate times, and all updates completed fine each time.

I have a hypothesis. The 4th generation of pies (including CM4) will use the older W11 release due to the hardware lacking the `atomics` CPU extension. (see the README for more details)
Something I have not tried on Pi4 hardware is Windows updates. If said updates do cause problems, this is likely the issue then, and if one of you finds a registry key that disables updates, I could add it to `autounattend.xml` so no manual workarounds are needed for fresh installs on the Pi4 and CM4.

GregE777, 2025-03-10:
Cm5 has recently become stable on these devices, I may just buy one and use the cm4 in my old Devterm. The extra performance might be needed for what I want to do anyway.

Botspot, 2025-03-10:
There must be another factor in play, because my pi 4 successfully completed Windows updates without issue. Part of the update procedure seemed to stop the windows RDP service, so in order to view the screen it seems I needed to use the Direct viewing mode. Not sure. I could not reproduce that issue later when back with using Remmina during the remainder of the update.

Do note that I manually triggered the update download in Settings, and restarted when prompted. Hoping this helps.

Botspot, 2025-03-10:
> I did see one thing I need to fix, I need to allocate more drive space next time. I have a 256GB card in this device and think I should have at least 60GB for the VM.

Easy fix. BVM has an `expand` mode that lets you add more space to the correct partition.