Botspot/bvm #62: Installation failing with error Windows 11 installation has failed

ujjaldey, 2025-03-11:
Hi,

Thanks for making this much-awaited project come true!

I am trying to install it on an RPi 5 with 16 GB RAM and a 64 GB SD card. I am trying the `bvm gui` option. But the installation fails every time with the error "Windows 11 installation has failed." I already tried 5 times, but no luck!

Screenshot:
[image]

Do you have any idea why the installation is failing and how to run it successfully?

Thanks in advance.

Botspot, 2025-03-11:
It's possible that a file permission was set wrong from an earlier run with `sudo`, or a downloaded file got corrupted in an earlier step.

When you are retrying the process, are you **deleting** the `~/win11` folder between tries and starting over with button 1, or are you just trying again with button 5?

ujjaldey, 2025-03-12:
Hi,  thanks for replying. 

I used the default user pi to execute all the steps. Tried `sudo` once, I believe bvm does not allow to run with `sudo`. 

Yes, before retrying I removed both `bvm` and `win11` directories. I even reinstalled the OS, but no luck.

Can I validate the checksum of the downloaded iso? I believe bvm checks that internally. 

Any help will be appreciated. Any chance of creating a custom image of the OS with the VM setup that can be instated directly on the RPi?

Botspot, 2025-03-12:
> Can I validate the checksum of the downloaded iso? I believe bvm checks that internally.

Sure. Let's check if your installer.iso has the same checksum as mine. (english united states)
```
sha1sum ~/win11/installer.iso
e2ee3254df969e2dc67623463e14384722ccf4d3  /home/pi/win11/installer.iso
```

> Any chance of creating a custom image of the OS with the VM setup that can be instated directly on the RPi?

That would technically be illegal for me to do.

ujjaldey, 2025-03-12:
Seems to be the same.

```
pi@raspberrypi:~/win11 $ sha1sum ./installer.iso
e2ee3254df969e2dc67623463e14384722ccf4d3  ./installer.iso
pi@raspberrypi:~/win11 $ 
```

Botspot, 2025-03-13:
> Seems to be the same.
> 
> ```
> pi@raspberrypi:~ $ cd win11
> pi@raspberrypi:~/win11 $ ls -lrt
> total 13382976
> -rw-r--r--  1 pi pi         442 Mar 11 22:22 connect.remmina
> -rw-r--r--  1 pi pi        4215 Mar 11 22:23 bvm-config
> -rw-r--r--  1 pi pi  5460387840 Mar 11 22:28 installer.iso
> drwxr-xr-x 15 pi pi        4096 Mar 11 22:29 unattended
> -rw-r--r--  1 pi pi    63700992 Mar 11 22:30 unattended.iso
> -rw-r--r--  1 pi pi           2 Mar 11 22:30 gui-steps-complete
> -rw-r--r--  1 pi pi 42960158720 Mar 11 23:45 disk.qcow2
> pi@raspberrypi:~/win11 $ sha1sum ./installer.iso
> e2ee3254df969e2dc67623463e14384722ccf4d3  ./installer.iso
> pi@raspberrypi:~/win11 $ 
> ```
> 
> Could it be an issue with disk space or RAM? I have 16GB RAM variant and 64GB SD card.
> 
> ```
> pi@raspberrypi:~/win11 $ df -h /
> Filesystem      Size  Used Avail Use% Mounted on
> /dev/mmcblk0p2   59G   19G   37G  35% /
> ```

You must have deleted the disk.qcow2 before running this command, because with disk.qcow2 present on `/`, your free space on your 64GB sd card would be very close to zero. that could be the issue, especially now with how QEMU expands the filesize gradually as Windows installs.

Try editing your bvm-config and changing the windows disk size from 40 to 30 GB.

ujjaldey, 2025-03-13:
Nope, even I noticed it before, but didn't have any explaination.

```
pi@raspberrypi:~ $ date
Thu 13 Mar 12:05:02 +08 2025
pi@raspberrypi:~ $ ls -lrth win11
total 12G
-rw-r--r--  1 pi pi  442 Mar 12 21:42 connect.remmina
-rw-r--r--  1 pi pi 4.2K Mar 12 21:42 bvm-config
-rw-r--r--  1 pi pi 5.1G Mar 12 21:48 installer.iso
drwxr-xr-x 15 pi pi 4.0K Mar 12 21:50 unattended
-rw-r--r--  1 pi pi  61M Mar 12 21:51 unattended.iso
-rw-r--r--  1 pi pi    2 Mar 12 21:51 gui-steps-complete
-rw-r--r--  1 pi pi  41G Mar 13 00:01 disk.qcow2
pi@raspberrypi:~ $ df -h /
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2   59G   18G   38G  32% /
pi@raspberrypi:~ $ date
Thu 13 Mar 12:05:56 +08 2025
pi@raspberrypi:~ $ 
```

Botspot, 2025-03-13:
Oh I think `ls` is showing the virtual size of the disk, rather than storage space actually used by the file.

ujjaldey, 2025-03-13:
No luck. It still failed at 57%.

ujjaldey, 2025-03-13:
@Botspot I finally managed to complete the installation successfully. The issue was with the SD card—using a new one resolved the problem. Thanks for your patience and all your help! You may close the ticket.