Botspot/bvm #63: Test at line 812 (sudo modprobe kvm) always fails if user has no sudo rights. kvm is built in to the kernel

thagrol, 2025-03-11:
Context: looking into running the VM as a user with no sudo rights.

Hardware: Pi5 4GB RAM

OS: RPiOS Bookworm with Desktop.

Line 812 tests for the presence of the kvm module by attempting to load it via `sudo modprobe kvm`. This fails if the user does not have sudo rights on `modprobe`.

On RPiOS Bookworm kvm is built in to the kernel so maybe test via something like `modinfo kvm | grep -q builtin` before attempting the modprobe.

Botspot, 2025-03-11:
It's not that simple unfortunately. Root is also necessary for:

- Installing dependencies
- Mounting the VirtIO .iso to extract drivers
- Mounting the disk.qcow2 to verify installation succeeded
- Removing Microsoft Defender
- USB passthrough
- Expanding the disk.qcow2

thagrol, 2025-03-11:
I'm sure it is more complicated and I should have been clearer about the exact circumstances. I'm also sure that some modes/subcommands of bvm won't be possible without elevated privileges.

I encountered this when running `bvm boot-nodisplay` from a user without sudo rights. Granting that user sudo rights on just /usr/sbin/modprobe made it go away.

I suggested the check via modinfo because as a kernel built in kvm does not appear in the output of lsmod and modprobe will only fail when run without sudo/root. It's failing due to a lack of rights not because "KVM kernel module is needed but missing" so the error message has strong potential to confuse or misdirect.

I don't currently have USB pass through enabled and no other errors were encountered or, at least, reported.

Botspot, 2025-03-11:
I have made that change now. https://github.com/Botspot/bvm/commit/08f9ec7e14b81aa6b560d94acb20f560a433f050
Be aware that if I ever add another dependency package, sudo-less usage will break. You may want to disable bvm updates in the config file to stay on a known working version.

closed 2025-03-11 by Botspot

thagrol, 2025-03-12:
Thanks, and yes, I had disabled updates.

commit 08f9ec7e (2025-03-11): Avoid running modprobe if kvm is builtin, to close #63