Botspot/bvm #64: partition

mtcgit09, 2025-03-12:
Is it possible to enlarge the partition created by default (40gb) or is it to late, so i have to start from scratch?
I am sure, that simply changing the BVM-config to a value >40 and enlarging the partition in windows doesn't work ...

I am using a NVMe with 1tb ;-)

Regards Mike

Botspot, 2025-03-13:
People still keep asking this, even after I added this feature to both GUI and CLI, mentioned it on the [readme](https://github.com/Botspot/bvm/blob/main/README.md#usage-instructions) and put it as the first answer on the [FAQ](https://github.com/Botspot/bvm/discussions/35). How odd...

But to answer your question, you should use the `expand` mode which asks how many gigabytes of space to add.

mtcgit09, 2025-03-13:
Sorry ...

Botspot, 2025-03-13:
No need to be sorry, I'm genuinely curious if I could have helped the available information be easier to understand, or if something specific confused you. :)

closed 2025-03-14 by mtcgit09