Botspot/bvm #65: Failed to expand filesystem to fill new space on 4th partition

Stanko33, 2025-03-12:
I've transferred my win11 folder to an external hard drive (and changed also the bvm/bvm shell file to match the new location where needed). As I change my current working directory in the terminal, to match the new win11/ path (going inside the win11 folder), I run `bmv/bmv expand ./`. The script then asks the number of gigabytes to expand the memory, and after some output, it says
"Failed to expand filesystem to fill new space on 4th partition"
Full output:
`win11 $ bvm expand ./
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██                                                      
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███                                                      
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█                                                      
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█                                                      
 █████🭠   🭖██🭡   █▉     🮋█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"The only thing more surprising than this working... is how well it works."
BVM mode expand
Enter number of gigabytes to ADD to the disk.qcow2 file: 1
Adding 1 GB to file... Done
Getting ready to work on partitions... Done
Expanding NTFS partition to take up new space... Done
Expanding filesystem...
ntfsresize v2022.10.3 (libntfs-3g)
Device name        : /dev/nbd0p4
NTFS volume version: 3.1
Cluster size       : 4096 bytes
Current volume size: 18789413376 bytes (18790 MB)
Current device size: 33821801984 bytes (33822 MB)
New volume size    : 33821798912 bytes (33822 MB)
Checking filesystem consistency ...
Cluster 1832527 is referenced multiple times!
Cluster 345621 is referenced multiple times!
Cluster 1832496 is referenced multiple times!
Cluster 1832497 is referenced multiple times!
Cluster 1832498 is referenced multiple times!
Cluster 1832499 is referenced multiple times!
Cluster 1832500 is referenced multiple times!
Cluster 1832501 is referenced multiple times!
Cluster 1832502 is referenced multiple times!
Cluster 1832503 is referenced multiple times!
100.00 percent completed
ERROR: Filesystem check failed!
ERROR: 117 clusters are referenced multiple times.
NTFS is inconsistent. Run chkdsk /f on Windows then reboot it TWICE!
The usage of the /f parameter is very IMPORTANT! No modification was
and will be made to NTFS by this software until it gets repaired.
/dev/nbd0 disconnected

Failed to expand filesystem to fill new space on 4th partition`

Can anyone solve this (probably a dumb problem I can't notice) ? By the way, very good job!

Botspot, 2025-03-13:
The disk image file seems to have already been corrupted somehow.
Did you copy it while windows was running, or did the VM encounter unsafe shutdowns or something?

Stanko33, 2025-03-13:
I did not copy it while the VM was running, buth I think it was because of an unsafe shutdown. In fact, after a couple of times re-running it, I managed to expand it. By the way, how can I increase the RAM memory of the VM? I've already installed More RAM from Pi-Apps.

Botspot, 2025-03-13:
> I did not copy it while the VM was running, buth I think it was because of an unsafe shutdown. In fact, after a couple of times re-running it, I managed to expand it.

It's good to know you got it working.

> By the way, how can I increase the RAM memory of the VM? I've already installed More RAM from Pi-Apps.

You can force a RAM size in the config file.

closed 2025-03-13 by Botspot