Botspot/bvm #66: Problems installing : mount: /tmp/tmp.NU8ybBLKKt: unknown filesystem type 'iso9660'

mrobinson-opi, 2025-03-13:
You should now be ready for the next step: ./bvm download /media/mrobinson/Downloads/win11
(base) mrobinson@orangepi5pro:~/bvm$ ./bvm download /media/mrobinson/Downloads/win11
 █████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██                                                      
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███                                                      
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█                                                      
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█                                                      
 █████🭠   🭖██🭡   █▉     🮋█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"Because nobody stopped to ask if this was a good idea. But clearly it is :)"
BVM mode download
Downloading Windows 11 ARM64 (English (United States))
  - Parsing download page: https://www.microsoft.com/en-us/software-download/windows11arm64
  - Getting Product edition ID: 3131
  - Permit Session ID: 7488964c-1328-475c-9873-ec9a3e9204df
  - Getting language SKU ID: 18624
  - Getting ISO download link...
  - URL: https://software.download.prss.microsoft.com/dbazure/Win11_24H2_English_Arm64.iso
/media/mrobinson/Do 100%[===================>]   5.08G  3.30MB/s    in 22m 59s 
2025-03-13 02:16:50 URL:https://software.download.prss.microsoft.com/dbazure/Win11_24H2_English_Arm64.iso?t=eb509378-a99b-43f5-bd8b-8c85c0342ce3&P1=1741917231&P2=601&P3=2&P4=ciWHFbS36GeasQ4yDg6uZ%2blpyTSpXjyka4nM0zjF0NguueRUBiIxMaZJLBMJZI%2fMsjnThIX%2fy%2f9xsxwGxLLZ0sox31rowEzcv%2fwcVNsM0M7UgxRbvigEaideF00aHN6teFQQELOa7OTU6eR55SttnfaGwpJxI%2bNWxP7pdyhHZKZi8K%2fVSWK%2bfBJYPAFmRcG8RIHz0XnWqXIsrlH%2b7sEjv%2f6JTT75possx5xJqeTpSu%2bx6G23UdcT47Fl6zd6LfRsaozzsgzdXJxtgmfNMMkfg%2b0efFVgo%2fGG4%2f4HwVFSlsWQi3fIN7SLk6Bm%2bQVJN8IkGrBN3y3afmD2GSGNT4410g%3d%3d [5460387840/5460387840] -> "/media/mrobinson/Downloads/win11/installer.iso" [1]
  - Verifying download... Done
  - Verification successful.
Modifying Windows ISO to boot to installer without keypress:
  - Searching for efisys_noprompt.bin to extract...
  - negative match at byte 5422041088
  - positive match at byte 5424738304
  - Extracting it...
  - Checksums match! Now searching ISO for sections to replace...
  - positive match at byte 1087488, replacing the next 1720320 bytes...
  - negative match at byte 5421072384, continuing to search...
  - positive match at byte 5423017984, replacing the next 1720320 bytes...
Done
Downloading virtio drivers
/media/mrobinson/Do 100%[===================>] 690.88M  3.43MB/s    in 4m 34s  
2025-03-13 02:21:59 URL:https://fedorapeople.org/groups/virt/virtio-win/direct-downloads/archive-virtio/virtio-win-0.1.266-1/virtio-win-0.1.266.iso [724434944/724434944] -> "/media/mrobinson/Downloads/win11/virtio-win.iso" [1]
Downloading Windows debloating script
Cloning into 'Win11Debloat'...
remote: Enumerating objects: 924, done.
remote: Counting objects: 100% (261/261), done.
remote: Compressing objects: 100% (84/84), done.
remote: Total 924 (delta 218), reused 179 (delta 177), pack-reused 663 (from 2)
Receiving objects: 100% (924/924), 294.80 KiB | 1.95 MiB/s, done.
Resolving deltas: 100% (672/672), done.
Extracting VirtIO drivers...
[sudo] password for mrobinson: 
mount: /tmp/tmp.NU8ybBLKKt: unknown filesystem type 'iso9660'.
       dmesg(1) may have more information after failed mount system call.
failed to mount virtio-win.iso for extracting files
(base) mrobinson@orangepi5pro:~/bvm$

Botspot, 2025-03-13:
Does this help? `sudo modprobe iso9660`

Run that, then repeat the same step and see if it succeeds this time.

mrobinson-opi, 2025-03-14:
modprobe: ERROR: could not insert 'isofs': Invalid argument

mrobinson-opi, 2025-03-14:
This is on an orange pi pro 16gb running armbian 6.1.99

Botspot, 2025-03-14:
Unless you upgraded the kernel and have not yet rebooted, your OS is to blame. Ask around for help to mount a ISO file, or find a better OS.

Botspot, 2025-04-06:
Were you able to get it working?