Botspot/bvm #68: Support for DietPi OS

fauxsocks, 2025-03-16:
Is or will there be support for DietPi OS? I was able to install Pi-Apps and bvm, however when launching the program, it seems nothing occurs. I do see loading by the mouse pointer for a few seconds, so it seems it does attempt a launch.

Botspot, 2025-03-18:
Run `ls -l /dev/kvm` and send back the output.

Botspot, 2025-04-06:
> crw-rw---- 1 root kvm 10, 232 Mar 16 01:24 /dev/kvm

The file permissions appear correct. I have the same file permissions on my system.

Try:
```
groups $USER
```
Maybe your user needs to be added to a group to be able to interact with kvm. This comes down to your OS and it's something you will need to find a solution for. I suggest you ask for help from DietPi forums, and if you do find a solution let me know and I can consider adding the fix within BVM so future users don't need to find a workaround.

jmaton, 2025-05-01:
Try this and let us know if you find success.

Add yourself to required groups "kvm" and "libvirt":
```
sudo usermod -aG kvm $(whoami)
sudo usermod -aG libvirt $(whoami)
sudo reboot
```

closed 2025-09-09 by Botspot