Botspot/bvm #69: Bridged Networking?

makrohard, 2025-03-17:
Great work, way better than fiddeling around with wine for my usecase, thank you very much!

I need the VM to be in the same network as the host. Can you give me a hint on how to enable bridged networking?

More precise, I need the VM to be able to communicate with the host via TCP/IP. If there is a different approach, that might work as well.

makrohard, 2025-03-17:
I haven't figured out yet, why, but I CAN talk to the host via TCP/IP. So, that works out-of-the-box. Thatfor: I'm closing the issue. Thanks again for your work, that makes my life so much easier!

closed 2025-03-17 by makrohard

Botspot, 2025-03-18:
Prettt sure the network is already bridged in the way that you are describing.

You can ask an AI for QEMU network flags that match your situation, then put them in the `network_flags` section of the config file.