Botspot/bvm #70: Windows won't boot up.

GeneralMuffenzz, 2025-03-21:
Every time I try to  boot up the OS I get the message  Failed to load Boot0001 UEFI Misc Device.

This started when I tried to sign in with my microsoft account when it was working, and after signing in it locked me out. So, I reinstalled the vm and this happened.

Botspot, 2025-03-21:
Please send screenshots or full error messages as seen in the terminal.

what device model, OS version, and storage device are you using?

GeneralMuffenzz, 2025-03-21:
BdsDxe: failed to load Boot0001 'UEFI Misc Device' from PciRoot (0x0) /Pci (0x7), (0x0) - Not Found 
I'm using raspbian version 12, have a rpi5 4gb, and I'm using a samsung sd card inside my pi.
Another detail, a windows shell opens up after around ten minutes but sits and does nothing, even after saying it will start running startup.exe

Botspot, 2025-03-21:
Someone else kept having windows bootup issues that were fixed by using a USB soldi state drive or other sd card, so you might want to try that.
You could also try installing windows again by deleting or renaming the disk.qcow2 file, then doing the firstboot step again.

GeneralMuffenzz, 2025-03-21:
It's probably unlikely that it's the sd card because it's worked before, so i'll try reinstalling windows.

GeneralMuffenzz, 2025-03-21:
It worked! Thanks for your help.