Botspot/bvm #72: BVM Will not boot

GamerPlayz-Gaming, 2025-03-24:
Hi, my bvm will not boot, it worked for 3 days then this morning it refused to boot, it would just say Qemu closed, maybe the OS failed to boot. It would say that over and over after about 20 minutes each attempt, I tried debugging and that did not work.

GamerPlayz-Gaming, 2025-03-24:
It just says

Waiting for RDP service on VM to start listening on port 3389... QEMU closed.
Consider supporting the developer Botspot, who is in a precarious life situation right now. Read more here:  https://github.com/sponsors/Botspot

QEMU process is no longer running. Maybe the OS failed to boot.
Operation failed! Please review the errors above.

GamerPlayz-Gaming, 2025-03-24:
I am trying to force a connection through the connect.remmina file

GamerPlayz-Gaming, 2025-03-24:
not working

Botspot, 2025-03-24:
Try booting with the direct mode instead of using an rdp connection.

GamerPlayz-Gaming, 2025-03-24:
When I do that it shows "Display Output Is Not Active"

Botspot, 2025-03-25:
Okay, check on task manager or `htop` to see if qemu uses the CPU, and if so, how long.

Btw, did Windows update the last time it successfully ran?

GamerPlayz-Gaming, 2025-03-25:
It never updated

GamerPlayz-Gaming, 2025-03-25:
And only bvm shows up on task manager

GamerPlayz-Gaming, 2025-03-25:
on htop it is showing that qemu is going on and off by 2 seconds each

GamerPlayz-Gaming, 2025-03-25:
I usually use the virtual machine for my ETCNomad Lighting Software

Botspot, 2025-03-25:
Try the mount feature to check for filesystem errors.
Also try `bvm boot-ramfb ~/win11` to get a lower level screen output, which could show a BIOS error screen.

GamerPlayz-Gaming, 2025-03-25:
This is what the hard drive says

BVM mode mount
The disk contains an unclean file system (0, 0).
Metadata kept in Windows cache, refused to mount.
Falling back to read-only mount because the NTFS partition is in an
unsafe state. Please resume and shutdown Windows fully (no hibernation
or fast restarting.)
Could not mount read-write, trying read-only
/home/etcadmin/win11/disk.qcow2 is now mounted to /media/etcadmin/bvm-mount3389
Press enter to unmount

GamerPlayz-Gaming, 2025-03-25:
Tried your command, got this

BVM mode boot-ramfb
qemu-system-aarch64: -drive file=/home/etcadmin/win11/disk.qcow2,if=virtio,discard=unmap,aio=threads,cache=none: Failed to get "write" lock
Is another process using the image [/home/etcadmin/win11/disk.qcow2]?
QEMU did not exit successfully. Code was 1

Botspot, 2025-03-25:
You need to press enter to unmount as it said, then try booting again with the boot-ramfb mode.

GamerPlayz-Gaming, 2025-03-25:
It is booting right now... I will say something if it goes wrong

GamerPlayz-Gaming, 2025-03-25:
How long will it take to go through the GUI boot?

Botspot, 2025-03-25:
> How long will it take to go through the GUI boot?

Could you post a screenshot of what you see?

GamerPlayz-Gaming, 2025-03-25:
I am using a raspberry pi, it cant do screenshots but I can try it from my phone

Botspot, 2025-03-25:
> I am using a raspberry pi, it cant do screenshots but I can try it from my phone

Yes it can, just press the printscreen key and then look in your pictures folder.
You can also install Flameshot from Pi-Apps to get an experience more like the windows snipping tool.

GamerPlayz-Gaming, 2025-03-25:
[image]

GamerPlayz-Gaming, 2025-03-25:
It has been stuck in this for about 30 minutes

GamerPlayz-Gaming, 2025-03-25:
I got a new screen

GamerPlayz-Gaming, 2025-03-25:
[image]

Botspot, 2025-03-25:
It's hard for me to help you if you don't describe or show me what is going on. What does the "new screen" say?

GamerPlayz-Gaming, 2025-03-25:
[image]

Botspot, 2025-03-25:
I don't think this is a problem with BVM, but with either Windows or with the quality of your sd card. Has your Pi or Windows experienced any unsafe shutdowns?

I suggest you research online for solutions, and maybe try pressing some of the keys Windows suggests. You could also mount the VM, copy your files out, and reinstall Windows.

GamerPlayz-Gaming, 2025-03-25:
It has not

GamerPlayz-Gaming, 2025-03-25:
Nevermind, I got them out

paymon23, 2025-03-30:
I can confirm reinstalling fixes this, TL;DR - don't update!!

closed 2025-04-06 by Botspot