Botspot/bvm #74: Failed to run on Orange Pi 5 Max

rwadeg, 2025-03-29:
It was working fine until the final restart. Then I got this message:

modprobe: FATAL: Module nbd not found in directory /lib/modules/6.1.43-rockchip-rk3588
find_available_nbd: 'sudo modprobe nbd' failed
mount: /media/orangepi/bvm-mount3389: special device p4 does not exist.
       dmesg(1) may have more information after failed mount system call.
mount_qcow2: mount command failed
It appears that Windows installation was not completed in disk.qcow2. Mounting the hard disk image failed.
Operation failed! Please review the errors above.

Botspot, 2025-03-29:
Your linux kernel may be missing the `nbd` kernel module. It depends on your distro.
The most I could do is make this an early requirement to warn users on incompatible systems at an earlier time.

Is there any chance you upgraded the kernel and have not rebooted yet?

Botspot, 2025-04-06:
Please respond, otherwise I will need to close the issue due to inactivity.

Botspot, 2025-04-11:
Closed due to inactivity. :(

closed 2025-04-11 by Botspot