Botspot/bvm #77: Windows instance turns off by itself?

paymon23, 2025-04-01:
Leaving it on and open overnight, I return and see that the instance has ended. 
How do I stop this from happening? I'm running Win 11 on Pi 5 with 4GB RAM.

Botspot, 2025-04-01:
This is either an issue with Windows, or the QEMU emulator, or your Pi, none of which I have precise control over. I have never heard of this happening, so more information is needed from you.

- Is it returning any errors in the terminal?
- Have you checked Windows log files?
- Have you tried using a screen recording tool such as mine to monitor the system all night to try to get a sense as to what is happening?

paymon23, 2025-04-01:
It's been behaving today, if it happens again I'll share any info above, thank you!