Botspot/bvm #84: OrangePi 5 Max - Ubuntu 22 LTS - Windows install step fails

spotopolis, 2025-04-08:
BVM installed and configured up to step 5 "install Windows 11"
When trying to install with verbose output enabled, I get the following error(s):

```
arm_release_ver: g13p0-01eac0, rk_so_ver: 10
qemu-system-aarch64: egl: eglBindAPI failed (core mode)
qemu-system-aarch64: OpenGL is not supported by the display
```

If I run `glxinfo -B`  I get the following error(s):

```
libGL error: glx: failed to create dri3 screen
libGL error: failed to load driver: rockchip
libGL error: gls: failed to create dri2 screen
libGL error: failed to load driver: rockchip
```

However, if I run `eglinfo`, I get what looks to be valid output:

```
EGL client extensions string:
    EGL_EXT_client_extensions EGL_EXT_platform_base
    EGL_KHR_client_get_all_proc_addresses EGL_KHR_platform_x11
    EGL_EXT_platform_x11 EGL_KHR_platform_gbm

GBM platform:
arm_release_ver: g13p0-01eac0, rk_so_ver: 10
eglinfo: eglInitialize failed

X11 platform:
EGL API version: 1.4
EGL vendor string: ARM
EGL version string: 1.4 Valhall-"g13p0-01eac0"
EGL client APIs: OpenGL_ES
EGL extensions string:
    EGL_NV_context_priority_realtime EGL_KHR_image_pixmap
    EGL_KHR_partial_update EGL_EXT_image_dma_buf_import
    EGL_KHR_config_attribs EGL_KHR_image EGL_KHR_image_base
    EGL_KHR_fence_sync EGL_KHR_wait_sync EGL_KHR_gl_colorspace
    EGL_KHR_get_all_proc_addresses EGL_IMG_context_priority
    EGL_KHR_no_config_context EGL_EXT_image_dma_buf_import
    EGL_EXT_image_dma_buf_import_modifiers EGL_EXT_yuv_surface
    EGL_EXT_pixel_format_float EGL_ARM_pixmap_multisample_discard
    EGL_ANDROID_native_fence_sync EGL_KHR_gl_texture_2D_image
    EGL_KHR_gl_renderbuffer_image EGL_KHR_create_context
    EGL_KHR_surfaceless_context EGL_KHR_gl_texture_cubemap_image
    EGL_EXT_image_gl_colorspace EGL_EXT_create_context_robustness

```

Botspot, 2025-04-20:
Are you remote connecting to the Orange Pi by any chance? I know that other people have successfully used BVM on that hardware, so either the issue is due to your OS or your specific setup.

nemo77com, 2025-04-21:
I have the same problems with my OrangePi 5 Max - Ubuntu 22 LTS:

◢◣ WARNING: Trying to use performance cores on your CPU only. 
PLEASE reach out to Botspot if this does not work.
Windows should install 100% automatically. This will take several hours.

arm_release_ver: g13p0-01eac0, rk_so_ver: 10
qemu-system-aarch64: egl: eglBindAPI failed (core mode)
qemu-system-aarch64: OpenGL is not supported by the display

QEMU did not exit successfully. Code was 1
Operation failed! Please review the errors above.

nemo77com, 2025-04-21:
RockChip RK3588 working: BVM is now [configured](https://github.com/Botspot/bvm/commit/79fa60a1be0a9b9c2c6599f77b1a1d799e82eba9) to use only the performance cores, making it functional on Rock Pi 5, Orange Pi 5, and many other single board computers that use this CPU. All ARM RockChip CPUs ought to work.

I try to install BVM via Pi-Apps

Botspot, 2025-04-21:
Same question for you, @nemo77com. Are you remote connecting to the Orange Pi by any chance? I know that other people have successfully used BVM on that hardware, so either the issue is due to your OS or your specific setup.

Botspot, 2025-04-23:
> [@Botspot](https://github.com/Botspot): Is a TeamViewer connection OK? When do you have time?

Teamviewer could very well be the entire problem. Try using your device connected to a physical screen and see if BVM works then.

nemo77com, 2025-04-24:
This is my system:
[image]

spotopolis, 2025-04-26:
I was using Orangepi5max_1.0.0_ubuntu_jammy_desktop_xfce_linux6.1.43.img from the official OPi5Max Google drive folder with the image running off of an m.2 nvme drive. That image had issues. As soon as I ran apt upgrade, the image broke and failed to boot. 

I switched to this Desktop Ubuntu build: https://joshua-riek.github.io/ubuntu-rockchip-download/boards/orangepi-5-max.html and I am able to now (mostly) install Windows.

If I try to use your GUI I get content GUI not responding errors asking for me to FC or wait for the application to respond. So I have just been running through the CLI commands to setup.

The system will make it through the WinPE section installing Windows, but I believe the scripted section that is supposed to shutdown the VM after the install to inject drivers fails and the VM just reboots instead skipping part of the unattend.xml as when it is going through the Windows first boot process configuring devices, it BSODs and then reboots once more. It then will make it to the "Welcome" animation, and ask for me to wait while it finishes the last steps, but it never moves past this screen. 

At this point I tried to shut down the VM after an hour, and when trying to boot the VM again, it could no longer find the boot device (virtual disk). 

I am currently going through the CLI setup again from scratch once more to see if it happens again and will report back either way.

spotopolis, 2025-04-26:
I was able to boot to the first run powershell script and then the system shutdown as designed. 

I could not get the system to boot with the "bvm/bvm boot ~/win11" command or "bvm/bvm boot-nodisplay ~/win11" command. The only way the system would boot again was with "bvm/bvm boot-ramfs ~/win11". When trying to boot any other way, it fails to find the boot device

spotopolis, 2025-05-14:
I tried once more to install from scratch and caught the BSODs during the second install screen. It BSOD twice and they were memory page faults. Then it still booted into windows and ran the powershell script then shutdown, however, as you can see in the second picture, the VM did not cleanly shutdown.

[image]

[image]