Botspot/bvm #85: QEMU pauses all by itself

GamerPlayz-Gaming, 2025-04-09:
When I launch the virtual machine through the **bvm boot-ramfb ~/win11** boot command, it would at random times auto pause itself, it happened to me in the past so I recreated a whole new vm and it fixed it until now and it keeps doing it when I boot it

GamerPlayz-Gaming, 2025-04-09:
I try to unpause it through the **Machine** tab but it just pauses itself again immediately

Botspot, 2025-04-09:
Strange. I have never heard of that happening before.
Could you upload a screen recording of it? Use botspot screen recorder

GamerPlayz-Gaming, 2025-04-09:
Nevermind, I fixed it, it was an issue with QEMU not the BVM, apparently if you don't have enough storage on the host computer, in this case it is the pi's os storage, for the bvm to run it's own files, then it will pause by itself so I cleared some stuff off of my computer and the bvm was working normally again

closed 2025-04-09 by GamerPlayz-Gaming