Botspot/bvm #88: Error when doing windows first run - Raspberry PI5 - 16gb board - Raspberry Pi OS Bookworm (from Raspberry PI imager)

curtterp-gh, 2025-04-19:
After preparing the Windows drive, I get an error when running the first run.

Board: RPI5
Memory: 16gb
OS: Bookworm
SD Storage: 512gb

Logged in User: Other user that has sudo privileges (Not the root user pi)

I did install xrdp so I can use Microsoft App (Formerly Windows RDP). This is how I access the PI, instead of putting a monitor/Keyboard/Mouse on the PI. 

The error sequence is: 

BVM mode firstboot
Windows should install 100% automatically. This will take several hours.
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI2: failed to authenticate
libEGL warning: DRI3: Screen seems not DRI3 capable
TU: error: ../src/freedreno/vulkan/tu_knl.cc:315: failed to open device /dev/dri/renderD128 (VK_ERROR_INCOMPATIBLE_DRIVER)
Opening /dev/dri/renderD128 failed: Permission denied
MESA: error: ZINK: failed to choose pdev
libEGL warning: egl: failed to create dri2 screen
qemu-system-aarch64: Could not access KVM kernel module: Permission denied
qemu-system-aarch64: failed to initialize kvm: Permission denied
QEMU did not exit successfully. Code was 1

/end of errors

Botspot, 2025-04-19:
> Logged in User: Other user that has sudo privileges (Not the root user pi)
> 
> I did install xrdp so I can use Microsoft App (Formerly Windows RDP). This is how I access the PI, instead of putting a monitor/Keyboard/Mouse on the PI.

These errors are definitely caused by at least one of those two ^ decisions you have made. It is likely that the new user you created is missing some permissions/groups that the `pi` user has by default. It is also likely that your choice to remote connect to a headless Pi is causing those DRI3 display errors.

Solution:

- Compare the output of `groups pi` with `groups <your new user>` and try adding your user to any missing groups.
- Also make your Pi output to a "physical" display - it does not need to actually be a screen, but add something to /boot/firmware/config.txt that is similar to this:
    ```
    # Set HDMI output to 1080p at 60Hz
    hdmi_group=1
    hdmi_mode=16
    
    # Enable HDMI audio
    hdmi_drive=2
    
    # Ensure HDMI is hotplugged
    hdmi_force_hotplug=1
    ```
    Then connect to that with VNC, or change your rdp server settings to use that "physical" display.

curtterp-gh, 2025-04-19:
Thank you for your response.

I was doing some research into the difference between VNC and RDP.

What I found is that RDP is creating a new instance, but VNC uses the current view (which is why VNC is better for helping someone to troubleshoot vs RDP)

I then enabled VNC on my Pi with raspi-config (through ssh), then downloaded Tiger-VNC to connect to the Pi.

I then started BVM (which, by the way, made me go through all the steps again)

When I got to the first boot step, there was no error, and Windows is installing now.

So the problem was between the chair and the keyboard. 

Once I get Windows installed, I will check to see if RDP is usable or if I need to keep using VNC.

So, since this issue was my own fault, I will close it now.

closed 2025-04-19 by curtterp-gh