Botspot/bvm #91: Issue: BVM becomes unusable via Remmina/other connection after unexpected VM shutdown

mcgdj, 2025-04-25:
If the VM gets shut down by accident, as shown in the title, there is no more possibility/practicality for it to be used with Remmina, as there is no background service that allows it to connect to a port.

This is a simple fix: You can add an option to restart the BVM background service when it has stopped (preferably before the boot options; after the QDisk config). If you can get this to work, I won't have to resort to a new image every time the VM shuts down. Obviously, this isn't @Botspot's doing - just a simple addition.

Additionally, I had issues with getting the direct connection to close. Not sure why that doesn't work, but it is probably my setup (just a few ol' generic input devices), because nobody else seems to be experiencing this. I'd suggest this be ignored for the moment, since as long as Remmina and others work with the VM, I see no use with a direct connection.

---
Anyway, I hope your financial situation works out. College is astonishingly expensive, but you have a gift for making useful, lightweight scripts in very little amounts of time. If you didn't make BVM, who would?

mcgdj, 2025-04-25:
Correct me if I'm wrong, but is the correct line to run in the background `bvm/bvm boot-nodisplay ~/win11`?

Botspot, 2025-04-25:
> If the VM gets shut down by accident, as shown in the title, there is no more possibility/practicality for it to be used with Remmina, as there is no background service that allows it to connect to a port.

That is very strange, as for the entirety of testing I always gave the VM an unsafe (instant) shutdown by killing QEMU. For me the RDP service, and by extension, remmina always worked flawlessly on next boot.
One time windows did report some data corruption, but it repaired itself automatically and continued working.

> Correct me if I'm wrong, but is the correct line to run in the background `bvm/bvm boot-nodisplay ~/win11`?

Correct.

mcgdj, 2025-04-25:
> > If the VM gets shut down by accident, as shown in the title, there is no more possibility/practicality for it to be used with Remmina, as there is no background service that allows it to connect to a port.
> 
> That is very strange, as for the entirety of testing I always gave the VM an unsafe (instant) shutdown by killing QEMU. For me the RDP service, and by extension, remmina always worked flawlessly on next boot. I even rebooted my Pi multiple times while the VM was still running, and never had any problems doing that either. One time windows did report some data corruption, but it repaired itself automatically and continued working.
> 
> > Correct me if I'm wrong, but is the correct line to run in the background `bvm/bvm boot-nodisplay ~/win11`?
> 
> Correct.

I'll have to get a screenshot then.
[I'll add it when I capture it]

Botspot, 2025-04-25:
> When I attempt to connect with Remmina, the shutdown triggers the background service to end (I... I think?) so when I attempt to reconnect the rdp service can't find the port.

The next time the VM boots, it should start the RDP service. If you boot direct (`boot-gtk`), that too should still start the RDP service. Maybe do that so you get a display output, then troubleshoot within Windows why the RDP service is seemingly not running.
I would also recommend doing a port scan on your local IP to check for the RDP service from the outside.

mcgdj, 2025-04-25:
> The next time the VM boots, it should start the RDP service. If you boot direct (`boot-gtk`), that too should still start the RDP service. Maybe do that so you get a display output, then [...]

Right, after I start the vm directly in any way things like remmina work.

Botspot, 2025-04-25:
> Right, after I start the vm directly in any way things like remmina work.

In that case, I'm confused. The boot button in the GUI simply runs windows with no display and later starts remmina. The behavior should be identical.
Please explain again in exactly what circumstances remmina never connects to your VM.

mcgdj, 2025-04-25:
> In that case, I'm confused. The boot button in the GUI simply runs windows with no display and later starts remmina. The behavior should be identical. Please explain again in exactly what [...]

If I shut down the VM, the service used to keep the machine running also stops. It is a very simple process to start it again, but it would require the user to enter a non-persistent command. On the user's end, they don't know what happened; their vm won't connect anymore. This is probably more of a feature request at this point, but it would _simplify_ things on the user's end to make sure the vm is running in the first place before attempting to connect, or have an option in the yad be to restart the vm background process.

Botspot, 2025-04-25:
Sounds like this is not an unusual issue with Windows then. (?)
I'm still unclear on what your end goal is here, but it sounds like you are using the GUI and want a way to start the VM again once it has been shutdown. I always just click the boot button again and it starts back up. So for that reason, I am having difficulty understanding what the problem is.

If you are doing something more complicated, like making the VM available for incoming connections on the local network, then I would recommend you avoid using the GUI for this. If this is what you are doing, then consider changing the windows registry to disable the shutdown button.
A more "hacky" solution would be to run bvm in the `boot-nodisplay` mode, inside a bash script that puts bvm inside a `while` loop, so that the VM starts again if the user initiates a shutdown.

closed 2025-04-25 by mcgdj

SeaDog58, 2025-05-08:
I have a similar problem probably related to how I have stopped a BVM session.  Now when I try "bvm/bvm boot ~/win11" I get an error message that says "Network port already in use.  BVM already running on system".  This appears even though I shutdown Debian Bookworm and started fresh.
How do you properly shut down a BVM session?  How do I close the 3389 port so I can proceed?

Botspot, 2025-05-09:
> I have a similar problem probably related to how I have stopped a BVM session. Now when I try "bvm/bvm boot ~/win11" I get an error message that says "Network port already in use. BVM already [...]

@SeaDog58 your problem is different, so please open a new issue and I would be happy to assist you with it.

SeaDog58, 2025-05-09:
Botspot,

I have solved my problem by going back to the installation steps starting with bvm/bvm new-vm ~/win11 and now everything works fine.

I think my problems arose from not closing the VM instance of Windows 11 properly.  I suggest you add to your instructions:

  *   Shutdown the VM instance of Windows 11 in standard manner by clicking on the Windows logo on task bar, then Power, and then Shutdown.
  *   Execute bvm stop in a terminal.

I intend to contribute to your efforts which are impressive.

John M. Poindexter, PhD
VADM, USN(ret)