Botspot/bvm #94: How to change NAT to Bridge in QEMU

ebcoelho, 2025-05-02:
I have a Pi5 HAT PCIe 2.5G network interface, since is not USB to pass-trough, how can I manage bvmconfig to change QMEU network from NAT to Bridge?  Can you add this feature on bvm gui?
I have been waiting since PI 3B to run Windows on RPI and now finally is true!
I'm planning to connect on Win11 using Guacamole for remote training using RPI5 with 16GB to run Window 11.
If you can help me to make this two features (qemu bridge/quacamole rdp) work..I can buy some coffee to help you to bring more features for this project!
Congratulations for your hard work!

ebcoelho, 2025-05-02:
Hello,

I want to give some update on my findings.

Looking inside bvm-config I was able to make rdp externally changing 127.0.0.1 to 0.0.0.0  without bridge.

Following the tutorial [Using AppleTalk with QEMU 9 on Raspberry Pi 5](https://biosrhythm.com/?p=2837) I was able to create br0 and enslave eth0  using the script br0-up.sh and the changes below on bvm-config
`network_flags=(-net nic,model=virtio,macaddr=52:54:12:34:56:78 -net bridge,br=br0)`

#Network settings: leave this alone unless you know what you are doing and need to change it.
**#For security, the VM's RDP port is only accessible on localhost, not on the LAN. To change that, replace 127.0.0.1 with 0.0.0.0**
#Do not change 3389 here with your value of rdp_port set above. This 3389 is the port on the Windows side that is redirected to $rdp_port, and will always be 3389.
#Comment this line out to prevent the VM from connecting to the Internet. (untested)
#network_flags=(-netdev user,id=nic,hostfwd=tcp:127.0.0.1:${rdp_port}-:3389 -device virtio-net-pci,netdev=nic)
`network_flags=(-netdev user,id=nic,hostfwd=tcp:**0.0.0.0**:${rdp_port}-:3389 -device virtio-net-pci,netdev=nic)`

Looking into [QEMU/Documentation/Networking](https://wiki.qemu.org/Documentation/Networking#The_legacy_-net_option)  I found that -net is deprecated and now they are using -netdev, so I changed the bvm-config and noted that tap0 interface now is attached to br0 when the Win11 is started and removed when stopped
`network_flags=(-device virtio-net-pci,netdev=nic,mac=52:54:12:34:56:78 -netdev bridge,id=nic,br=br0)`

I was not able to get hostfw and bridge running on same interface to use 'bvm/bvm connect /home/training/win11' for RDP connection but the workaround I found was to add 2 network interfaces: one for rdp that creates a NAT interface with DHCP server and another for bridge but I got 2 default gateways and I don't know how this gonna work in production.
`network_flags=(-netdev user,id=nic0,hostfwd=tcp:127.0.0.1:${rdp_port}-:3389 -device virtio-net-pci,netdev=nic0 -net nic,model=virtio,macaddr=52:54:12:34:56:78 -net bridge,br=br0)`

I got a small issue on this scenario, after sometime eth0 gets his original ip and the bridge the same because the script br0-up.sh essentially copies the ip from eth0 and assign to br0. I still have to figure out if it is the dhcp lease time or NetworkManager trying to take over the eth0.

The goal now is to use 'bvm/bvm boot-nodisplay /home/user/win11' and  'bvm/bvm connect /home/training/win11' but I am using bridge and no hostfwd is managed so when I run for rdp I get the message below:
`Waiting for RDP service on VM to start listening on port 3389... `

ebcoelho, 2025-05-09:
After testing armbian and ubuntu with raspberry pi 5, raspian with debian 12 bookworm worked better with no errors when running BVM. 

After reflashing the SD card with raspian, I installed BVM and faced the error message: **failed to parse default acl file `/etc/qemu/bridge.conf'** when trying to use bridge interface changing bvm-config. 

Looking into how to [Configuring QEMU bridge helper after “access denied by acl file” error](https://blog.christophersmart.com/2016/08/31/configuring-qemu-bridge-helper-after-access-denied-by-acl-file-error/) I created the /etc/qemu/bridge.conf and added the parameter `allow all`:
```sudo mkdir -p /etc/qemu
echo "allow all" | sudo tee /etc/qemu/bridge.conf```

After the fix above, the error message changed to **failed to create tun device: Operation not permitted**. After doing more resource I found this article [qemu kvm: operation not permitted fix](https://danielms.site/zet/2024/qemu-kvm-operation-not-permitted-fix/) showing how to fix permissions of qemu-bridge-helper:
`sudo chmod u+s /usr/lib/qemu/qemu-bridge-helper`

On network manager, I removed eth0 and created and slaved on bridge br0 that showed to be more stable working with BVM and also virt-manager running alpine with cockpit using bridge also at same time.

I found a project called [qemu-rdp](https://lists.libvirt.org/archives/list/devel@lists.libvirt.org/message/WHHKRS5FF6YIRDFXUJHOP2PEURBDJCEW/) to use libvirt that works with cockpit to access the console using rdp. The idea to get the console using rdp from qemu/kvm directly as the network interface of win11 will be in bridge to not loose connectivity if the user makes some bad configuration enabling 801.x for example. We have Spice and VNC using virt-manager or cockpit. Getting the remote console of win11 over RDP directly or using cock-pit over web will be a nice feature.

Botspot, 2025-09-09:
Thank you for leaving behind instructions for how to do this, in case there is anyone else who has a similar request. I've linked this issue in #35.

closed 2025-09-09 by Botspot

Lightwel, 2025-11-07:
Hello Coelho

I am trying to bridge BVM back to host for two-way traffic.   I'm a bit confused by how many steps are in your post: 

Are these all the steps after creating br0 enslaved to eth0?

1   change network flags in bvm-config to read (I want RDP accessible from a remote machine)

network_flags=(-netdev user,id=nic0,hostfwd=tcp:0.0.0.0:${rdp_port}-:3389 -device virtio-net-pci,netdev=nic0 -net nic,model=virtio,macaddr=52:54:12:34:56:78 -net bridge,br=br0)

2    create the file /etc/qemu/bridge.conf and add the text  "allow all":

and

3   chmod /etc/qemu/bridge.conf to u+s 

?

Many thanks
R