Botspot/bvm #95: "Run Windows 11" option problem

Animalmicraft, 2025-05-04:
The system installed fine, but nothing happened. When I tried to boot it again, the run option didn't appear unlocked. I could only get in by hitting install windows and going into the uefi and booting it from there. The problem is that it's tedious and doesn't let me use a USB drive. How do I fix this?

Botspot, 2025-05-04:
Thanks for opening an issue. I have an idea for why the "run Windows" button wwas not unlocked. Did you encounter an error during installation saying it did not seem to be fully installed? Or was the process interrupted in some other way, like shutting down your pi after windows installed, without shutting down the VM first?

Well regardless of the cause, if you want to unlock the "Run Windows 11" button, it's easy. Just go to the win11 folder, find the `gui-steps-complete` file, and change the value to 5. Save the file and now all buttons should be unlocked.

Animalmicraft, 2025-05-04:
The PI froze, but I had already installed it. When it came back to normal, it told me the installation had failed, but editing the file worked for me. Thank you very much.