Botspot/bvm #96: Use Windows without spending or using Wi-Fi

Animalmicraft, 2025-05-04:
Sorry for reopening a discussion, but I'd like to know how to use Windows with Remmina (to be able to transfer files, Wi-Fi, audio, etc.) without using Wi-Fi.

Botspot, 2025-05-04:
If the VM and Remmina are both running on the same device, the RDP internet traffic should stay local to your device.
What are you observing that would make you think otherwise?

Animalmicraft, 2025-05-04:
When I disconnect the wifi the machine doesn't work (I don't know if it's normal), my internet slows down, it works even better with direct but I would like to use copy and paste and other things that only work with Remmina. Could it be because my internet is slow?

Botspot, 2025-05-04:
It should work without any internet connection. I just tested this right now and it worked fine with Remmina.
Could you share more details about what happens when you try this?

Animalmicraft, 2025-05-04:
█████🭏 🭖█🭀  🭋█🭡 ██◣   ◢██                                                      
 █▊  🭨█ 🭦█🭐  🭅█🭛 ███◣ ◢███                                                      
 █████🭪  🭖█🭀🭋█🭡  █▉◥█🭩█◤🮋█                                                      
 █▊  🭨█  🭦█🭐🭅█🭛  █▉ ◥█◤ 🮋█                                                      
 █████🭠   🭖██🭡   █▉     🮋█                                                      
BOTSPOT  VIRTUAL  MACHINE                                                       
"Your Pi is about to feel so sophisticated. Or nauseated. Maybe both. Not sure."
fatal: no es posible acceder a 'https://github.com/Botspot/bvm/': Could not resolve host: github.com
BVM mode boot-nodisplay
fatal: no es posible acceder a 'https://github.com/Botspot/bvm/': Could not resolve host: github.com
BVM mode connect-remmina
Waiting for RDP service on VM to start listening on port 3389... mass_dns: warning: Unable to determine any DNS servers. Reverse DNS is disabled. Try using --system-dns or specify valid servers with --dns-servers
mass_dns: warning: Unable to determine any DNS servers. Reverse DNS is disabled. Try using --system-dns or specify valid servers with --dns-servers
mass_dns: warning: Unable to determine any DNS servers. Reverse DNS is disabled. Try using --system-dns or specify valid servers with --dns-servers
mass_dns: warning: Unable to determine any DNS servers. Reverse DNS is disabled. Try using --system-dns or specify valid servers with --dns-servers
mass_dns: warning: Unable to determine any DNS servers. Reverse DNS is disabled. Try using --system-dns or specify valid servers with --dns-servers

from there he keeps saying "mass_dns: warning: Unable to determine any DNS servers. Reverse DNS is disabled. Try using --system-dns or specify valid servers with --dns-servers" and it doesn't stop. Do I have to wait longer?
And by the way, does this use up internet?

Botspot, 2025-05-04:
`mass_dns: warning: Unable to determine any DNS servers. Reverse DNS is disabled. Try using --system-dns or specify valid servers with --dns-servers` is a harmless error which I fixed today after testing this for you. It should work.
Observe your CPU usage to see if the VM is booting in the background or not, and maybe wait longer.

Animalmicraft, 2025-05-04:
It finally worked for me, trying your app, it's incredible, the only thing missing is a shared ram and that I bought a 16GB ram raspberry xd

closed 2025-05-04 by Animalmicraft

Botspot, 2025-05-04:
> the only thing missing is a shared ram and that I bought a 16GB ram raspberry xd

RAM is shared between linux and windows, so is that not what you meant?

Animalmicraft, 2025-05-04:
I mean that even in the virtual machine I see 8GB of ram that the raspberry has, and that the virtual machine spends everything it can until all the ram is full, the problem I have had when using qemu is that the selected amount of ram is filled at once and that is why 2 GB are always left for our main system, normally something that only the experts in qemu would know how to solve. What I am referring to is not an error, it is a suggestion in case too much ram is used, and if there is an opportunity to solve it.

Botspot, 2025-05-04:
> I mean that even in the virtual machine I see 8GB of ram that the raspberry has, and that the virtual machine spends everything it can until all the ram is full, the problem I have had when using [...]

I think I understand you. BVM is unique in how it dynamically scales the VM's ram usage. Try it and see. Use a RAM-hungry application in linux and watch as qemu compensates by using less ram. I have designed BVM to give the VM as much ram as possible, but it watches free ram and always maintains a certain amount free, which can be configured. It gives up ram as needed in order to share fairly.

Animalmicraft, 2025-05-04:
This is the application I've been waiting for since my pi3.