Botspot/bvm #97: BVM (Win11) on Pi 5 won't detect Slamtec RPLidar-C1 plugged into any USB port

CrosswordMike, 2025-05-04:
Installed BVM (Win11) on Raspberry Pi 5. Win11 is running fine, but won't recognize device plugged into any USB port. 64-bit driver from Silicon Labs has been installed.
When device is plugged into my laptop, it appears on the Device list with the driver under 'Ports'. No Ports appear on the Device list under BVM. Also not listed under USB Connectors. And 'Scan for Hardware Changes' does not detect it.

Botspot, 2025-05-04:
I assume you did, but did you make sure to configure BVM to passthrough that USB device to the VM?
My other thought is that maybe you need a USB driver compatible with ARM Windows, which may or may not exist. A normal 64-bit driver intended for x86_64 CPUs won't work.

CrosswordMike, 2025-05-04:
I did configure the passthrough. I'll see if I can find an ARM-compatible driver. Thx

closed 2025-09-09 by Botspot