Botspot/bvm #98: "password has expired"

Botspot, 2025-05-04:
Hey everyone. Sorry about this issue. I was not aware that by default, Windows requires you to change your password every 42 days for local accounts. This effectively breaks RDP connections (Remmina and FreeRDP) until you change the password, which you cannot easily do with RDP connections being declined.

To fix:

1. Boot the VM without using RDP. Boot with the "Direct" view mode in GUI, or `~/bvm/bvm boot ~/win11` in a terminal.
    You should see this screen.
    [image]
2. Change the password as prompted.
3. Edit BVM's config file to tell it the new password. The file is named `bvm-config` and it's stored in the win11 folder. (unless you named it something else)
    Now, RDP connections with Remmina/FreeRDP should work again.
4. To prevent your new password from expiring 42 days from now, open an administrator command prompt in Windows, and run this command: `wmic UserAccount set PasswordExpires=False`    Full disclosure: Until 42 more days, I cannot 100% guarantee this works, but [this site](https://www.tenforums.com/tutorials/87379-enable-disable-password-expiration-local-accounts-windows-10-a.html) says it works.
5. Continue using BVM as before.

I [have added](https://github.com/Botspot/bvm/commit/6568fb5368b6ab04fb87f014ded001fc770b793f) `wmic UserAccount set PasswordExpires=False` to the list of commands that runs in Windows during its installation, but that change will only help new BVM users who are installing Windows after today. Everyone else needs to follow the workaround above.

Sorry for the inconvenience. Comment below if you need further help.
-Botspot

jmaton, 2025-05-09:
Another option might be to run this in windows console:

`wmic UserAccount where Name="USERNAME" set PasswordExpires=False`

Botspot, 2025-05-29:
> Another option might be to run this in windows console:
> 
> `wmic UserAccount where Name="USERNAME" set PasswordExpires=False`

wmic command has been deprecated in latest Win11. That won't work.

chivarpilones, 2025-06-11:
hi botspot,
 I have followed the steps above but on the 2nd image after I enetered default pass win11arm and provide my new password win12arm twice to confirm I get the Access Denied prompt

Botspot, 2025-08-27:
> When I select Direct after getting the "Password Expired" error, the startup window appears with my name and icon and a Start to click. When I click Start, there are a few seconds of whirring, then the window resets to the startup window with nothing happening. I try changing my username and pasword (including nulls) in the config file to no avail. I believe I was using a PIN to sign in to Windows 11. I do have a Windows username and password, but there is no way to enter them. What do I do? Have I lost this Windows 11 installation?

First, boot `direct` and set a new password. Remember that password. Not, booting `direct` you should be able to enter that password manually and login.
To use RDP connections, just put that password into the config file. The config file does not change your windows password.

Botspot, 2025-08-27:
@DJenner8 Your issue is a windows issue, and windows is something I have little experience in. Maybe try booting windows in some sort of recovery mode, or do that hack trick where you replace the magnifier exe with the command prompt exe (using the bvm mount function) and then run magnifier from the ease of access toolbar in the lockscreen. Then once you have command prompt, you could probably fix whatever is going wrong. I would recommend explaining the situation to an AI and ask for guided help. https://duck.ai is a good resource for this.