Botspot/bvm #104: win11 can't boot after finish the win11 installing

udhgu, 2025-05-24:
qemu-system-aarch64: -netdev user,id=nic,hortio-net-pci,netdev=nic: warning: short-form boolean option 'hortio-net-pci' deprecated
Please use hortio-net-pci=on instead
qemu-system-aarch64: -netdev user,id=nic,hortio-net-pci,netdev=nic: Invalid parameter 'hortio-net-pci'
QEMU did not exit successfully. Code was 1
BVM mode connect-remmina
QEMU does not appear to be running. You need to run the VM first before you can view its screen.
Operation failed! Please review the errors above.

udhgu, 2025-05-24:
it can display normally during the installation process of 11

udhgu, 2025-05-25:
the problem has been solved

closed 2025-05-25 by udhgu

Botspot, 2025-05-25:
What did you do to solve it? Could BVM have done anything more helpful here to make the process smoother for you>