Botspot/bvm #117: Activating USB passthrough breaks installation

SuperCowProducts, 2025-07-20:
Hello, I've had lots of fun running BVM, until I tried using USB passthrough functionality - now even every time I boot BVM 'normally' (`/home/libird/.local/share/bvm/bvm boot-nodisplay ~/win11 &> "$(date '+%d-%m-%Y.%H;%M;%S')"`), my whole system goes bonkers, and I'm forced to perform hard reboot with CTRL-ALT-PRNT + R E I S U B! Login shells in different ttys fail to appear, including inside the Konsole app, where every new tab displays "Warning: Could not find an interactive shell to start." (I start up Plasma from Raspberry Pi OS Lite with `startplasmamobile`), and new apps get stuck while loading, probably because they aren't able to spawn new processes.

Now all of this seems quite enigmatic, so I have attached `journalctl` logs as well as verbose output derived from the starting command I specified earlier.

To be more precise and to better isolate bugs from broken system configurations, I usually perform a hard reset of the software I'm testing, but becaue restarting downloads and installs of Windows is a very tedious process I was wondering @Botspot is there some command I can run to do that without touching those files? Especially because what seems to be broken here is something linked to the interface (Remmina / USB / mounting?) rather than Windows 11 itself, which was working beautifully before): otherwise why else would my entire system be impacted 🤔...

I'd be happy to test this on a fresh install of Raspberry Pi OS Desktop (Raspberry Pi 5 8GB), although because I've encountered similar problems on the 'official' OS as well I'm tempted to believe this is a bug unrelated to using Plasma.

I was also thinking of using USB boot as a workaround, but after that failed I realised the most useful move for me and the community at the moment was focus on fixing the tool that is being actively developed (which is also I do believe the coolest 😎).

[20-07-2025.20;42;01-journalctl.txt](https://github.com/user-attachments/files/21338306/20-07-2025.20.42.01-journalctl.txt)

[20-07-2025.19;55;01.txt](https://github.com/user-attachments/files/21338307/20-07-2025.19.55.01.txt)

SuperCowProducts, 2025-07-20:
Just wanted to add this is the most important feature for me at the moment, much more than WiFi, as sadly I have to use software for recovering my phone which only runs on Windows 😿,

SuperCowProducts, 2025-07-20:
I'm currently repeating the BVM installation process on a freshly-flashed USB: if only there was something like SD Card Copier only I can store the starting 'SD Card' virtually as a backup: that could be an effective way to roll back, unless anyone is aware of programs that work well for this on the Pi (Open Suse has this built in, and I heard on Arch it's quite common, but I wouldn't be so sure for RpiOS / Debian Stable...

Botspot, 2025-07-21:
> Now all of this seems quite enigmatic, so I have attached `journalctl` logs as well as verbose output derived from the starting command I specified earlier.

Your verbose output from running bvm seems to be truncated, and the only obvious errors I saw in journalctl had to do with `obsidian` crashing. Was there anything else in those files that you wanted me to see?

> To be more precise and to better isolate bugs from broken system configurations, I usually perform a hard reset of the software I'm testing, but becaue restarting downloads and installs of Windows [...]

Yes, it's easy to keep the windows files intact while reverting configuration changes. Simply copy the `bvm-config` file from the `bvm` folder to your `~/win11` folder. The USB-forwarding config is stored in there and preserved across boots, so clearing that out should return you to the config you had before.

My best guess is that perhaps by forwarding a USB device, it caused misbehavior in other processes in your OS. In order to give the VM access to the device, the rest of Linux userland *loses* access to the device for as long as QEMU is running.

> I'd be happy to test this on a fresh install of Raspberry Pi OS Desktop (Raspberry Pi 5 8GB), although because I've encountered similar problems on the 'official' OS as well I'm tempted to believe this is a bug unrelated to using Plasma.

Yes, please test on pi os desktop. If any issues still arise, it's likely due to your hardware setup, and/or the USB devices you have chosen to forward.

> I was also thinking of using USB boot as a workaround, but after that failed I realised the most useful move for me and the community at the moment was focus on fixing the tool that is being actively developed (which is also I do believe the coolest 😎).

Correct. Thanks for opening this issue, and with any luck it will be an easy fix, either for the code, or for the documentation. (If it turns out that forwarding a specific USB device was causing these issues)

SuperCowProducts, 2025-07-21:
So, I got the OS to boot and everything (yay ;) ), and tried activating USB passthrough: after installing an app I downloaded which I needed for fixing my phone using `wget "https://en-us.support.motorola.com/app/answers/detail/a_id/168095" -O "Software_Fix_v7.3.4.13_setup.exe"` (took a reboot to actually get into the useful part of the app, as the system froze the first 'round'), I noticed that although the USB passthrough value had been uncommented in `$HOME/win11/bvm.config`, on Windows in the bottom right corner or in the Motorola App itself there was no sign of my device, while running `fastboot devices` on Raspberry Pi OS revealed my device no problem (I'm sure it's supposed to be the other way round, where the device gets 'sent' to Windows and instead `fastboot devices` output on Raspberry Pi OS should be empty as a consequence).

After a shutdown from Windows GUI my OS no longer booted again, here are some log files are collected (which should be more useful than the first ones <i

Botspot, 2025-08-30:
As of commit https://github.com/Botspot/bvm/commit/6355505d31b3fb15baa4c5d48bedf17457276fa4, USB passthrough is now working more reliably. After BVM updates, could you retry this and see if it fixed at least some of your issues?