Botspot/bvm #120: Headless RPI doesn't play well with RDP Port 3389

Botspot, 2025-07-23:
### Discussed in https://github.com/Botspot/bvm/discussions/119

<div type='discussions-op-text'>

<sup>Originally posted by **Jamie-A-Taylor** July 23, 2025</sup>
I am trying to run bvm from a windows host machine, where I ssh into rpi and then run the bvm boot-nodisplay command.

looking to then be able to rdp from the host to the qemu session

I have modified the port in the bvm config to try and resolve the rdp port conflict but this hasn't helped

the error I get each time I try to boot either normal boot or boot-no display is "network port 3389 is already in use" is this because I am already rdp into the pi?</div>

Moving this to an issue, as it's more suited here than a discussion.

Jamie-A-Taylor, 2025-07-23:
@Botspot 

Thank you for your reply!

XRDP service was causing the port collision on 3389.

I was also modifying the bvm/bvm-config instead of the win11/bvm-comfig.

So I now connect to the pi via ssh to avoid the collision of the rdp server. 

When I run bvm/bvm boot ~/win11
Terminal returns 

BVM needs a desktop environment to run boot.

Does it matter which GUI is installed currently I am using xfce4?

When I run bvm/bvm boot-nodisplay ~/win11
Terminal returns

Qemu-system-aarch64: could not access kvm:permission denied
Qemu-system-aarch64: failed to initialise kvm:permission denied
Qemu did not exit successfully code was 1

Botspot, 2025-07-23:
I think it's worth your time to familiarize yourself with the documentation a bit more. For example, the config file itself warns to not edit it if it's in the bvm folder...

> BVM needs a desktop environment to run boot.

In this case, it does not seem like you want the VM to boot in a graphical window on linux, it sounds like you just want to start a VM with a RDP server, and run it on a headless pi.
So `boot-nodisplay` is the correct mode you should use.

Also you will need to allow incoming connections from the local network, as that is disabled by default, but doing that is explained in the config file.

closed 2025-08-26 by Botspot